uunicorn/python-validity #9: Pair succeeds, open9x() fails on 06cb:009a

RaghavRao, 2020-07-07:
Hi, I'm not sure if I'm doing this right:

```

$ python3 pair.py

Initializing flash...
Detected Flash IC: W25Q80B, 1048576 bytes
Sleeping...
Clean slate
Uploading firmware...
Sensor: 57K0 FM-3367-003   
Loaded FWExt version 1.2 (Mon Jan 27 01:32:09 2020), 8 modules
Sleeping...
Calibrating...
Opening sensor: 57K0 FM-3367-003   
Warning: no calibration data was loaded. Consider calibrating the sensor.
Calibration iteration 0...
Ignoring spurious interrupt: 0000000000
Calibration iteration 1...
Ignoring spurious interrupt: 0000000000
Calibration iteration 2...
Ignoring spurious interrupt: 0000000000
Requesting a blank image...
Ignoring spurious interrupt: 0000000000
Init database...
Creating a new user storage object
Creating a host machine GUID record
That's it, pairing's finished
[root@x1-yoga-3rd python-validity]# python3
Python 3.8.3 (default, May 15 2020, 00:00:00) 
[GCC 10.1.1 20200507 (Red Hat 10.1.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from prototype import *
>>> open9x()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/raghav/Development/python-validity/proto9x/init.py", line 22, in open
    usb.send_init()
  File "/home/raghav/Development/python-validity/proto9x/usb.py", line 57, in send_init
    assert_status(self.cmd(unhexlify('01')))
  File "/home/raghav/Development/python-validity/proto9x/util.py", line 9, in assert_status
    raise Exception('Failed: %04x' % s)
Exception: Failed: 0315

```

uunicorn, 2020-07-07:
Thanks. Looks like calibration logic lacks the reboot() in the end. Can you close this interactive python session and try again?

RaghavRao, 2020-07-08:
Enrolling suceeded after exiting the session, opening again, and running the two lines again:

```
>>> from prototype import *
>>> open9x()

```

uunicorn, 2020-07-08:
I've updated `pair.py` to do a `reboot()` before exit, so should not be a problem.

closed 2020-07-09 by RaghavRao