uunicorn/python-validity #14: Exception: Failed: 044f

Titaniumtown, 2020-07-13:
I'm on arch linux, using the AUR package.
```
Python 3.8.3 (default, May 17 2020, 18:15:42)
[GCC 10.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from prototype import *
>>> open9x()
Flash has 5 partitions.
Clean slate
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/site-packages/validitysensor/init.py", line 43, in open
    open_common()
  File "/usr/lib/python3.8/site-packages/validitysensor/init.py", line 30, in open_common
    upload_fwext()
  File "/usr/lib/python3.8/site-packages/validitysensor/upload_fwext.py", line 26, in upload_fwext
    fwi=get_fw_info(2)
  File "/usr/lib/python3.8/site-packages/validitysensor/flash.py", line 80, in get_fw_info
    assert_status(rsp)
  File "/usr/lib/python3.8/site-packages/validitysensor/util.py", line 9, in assert_status
    raise Exception('Failed: %04x' % s)
Exception: Failed: 044f
```

uunicorn, 2020-07-13:
Can you enable tracing and try again?

uunicorn, 2020-07-13:
I'm pretty sure 044f means "signature failed".  Could it be that the previous firmware upload was interrupted? Can you try again after doing factory reset?

Titaniumtown, 2020-07-13:
I did a factory reset, now i get the error: 
```
Flash was not initialized yet. Formatting...
Detected Flash IC: W25Q80B, 1048576 bytes
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/site-packages/validitysensor/init.py", line 43, in open
    open_common()
  File "/usr/lib/python3.8/site-packages/validitysensor/init.py", line 26, in open_common
    init_flash()
  File "/usr/lib/python3.8/site-packages/validitysensor/init_flash.py", line 158, in init_flash
    reboot()
  File "/usr/lib/python3.8/site-packages/validitysensor/sensor.py", line 63, in reboot
    raise RebootException()
validitysensor.sensor.RebootException
```

Titaniumtown, 2020-07-13:
Nevermind, it works now :)

closed 2020-07-13 by Titaniumtown

uunicorn, 2020-07-13:
That was not really an error. The pairing process is iterative. After each step the device is expected to reboot and disconnect from the host. I'm using RebootException to bail out and indicate that device has restarted and needs to be reopened. dbus-service is doing it automatically, but when you're using low-level tools, you have to do restart them manually.