uunicorn/python-validity #15: Factory Reset needed after booting into Windows

RaghavRao, 2020-07-13:
This is probably unavoidable but it looks like windows claims the device causing python-validity to fail with the DB GUID mismatch. Running factory-reset, restarting the service and re-enrolling works.

Is there a work-around so that the user doesn't have to re-enroll on every boot to windows? Such as backing up the enrolled finger, silently factory-resetting and restoring the fingers if:

1) fprintd-enroll has been used on this machine before succesfully
2) fprintd-delete has not been called
3) Only right index finger is enrolled?

Or maybe for now we can catch the DB exception and log just a message saying factory-reset needs to be run.

uunicorn, 2020-07-16:
I think we've discussed it a little bit in #3 . The problem is - we need to have a pragmatic balance between security and compatibility with Windows. Right now there is a workaround - boot in Windows and get the Machine GUID (HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Cryptography\MachineGuid). Boot in Linux and change the hardcoded value in `/usr/lib/python3/dist-packages/validitysensor/init_db.py`. In future we should probably keep a file like `/usr/share/python-validity/my-guids.txt` which will contain a list of GUIDs we want to work with. If the device is paired but the GUID is not in the file, the driver should ignore such device, however the package configuration should prompt the user if they trust the device and if the user's answers is "yes" - it should add this GUID to the file, if the answer is "no", we should force a factory reset which will create a new GUID and append it to the same file.

RaghavRao, 2020-07-17:
I really like this idea. I can implement it this weekend if you like, maybe on a separate branch.
Would you prefer to keep the discussion on #3?

uunicorn, 2020-07-17:
> I really like this idea. I can implement it this weekend if you like, maybe on a separate branch.

I have doubts. I'm not really sure how much the GUID feature is contributing to the overall device security. It's not like MachineGuid is a secret. Moreover, the device itself allows anyone to read and modify this GUID record without performing a factory reset. So, in theory you can take a genuine Validity sensor, pair it using my MachineGuid, product name and serial number and plug it into my computer. In this case the Windows driver on my computer will never attempt to factory reset your device and it will use your finger templates to try and log you in on my computer (it won't work even if you guess my SIDs, but that's another story). So, GUID is probably just a convenient way to get rid of enrolled users after you reinstall Windows. There is not much value in keeping the template database with SIDs for accounts which no loner exist.

I think as part of establishing the trust, instead of storing a GUID we should probably store a public key of the device and ignore the GUID record completely. We may not even create one as part of the pairing. This way Windows won't force a factory reset if you initially paired with Linux.

The device key I mentioned is generated as part of the pairing process. There is currently no way to extract a secret part of it unless someone has tampered with a device before the pairing. This will keep us safe from a hardware clone attack (which is now possible due to [leaked Validity keys](https://www.synaptics.com/sites/default/files/fingerprint-sensor-VFS7500-security-brief-2020-07-14.pdf)).

So, the approach is sort of the same: 
- when a blank device is plugged in: silently pair with it and store the device's public key in a local `known_devices` file
- when a paired device is plugged in: ask the user if she trusts this device. Options are: 
  - **Yes**: store the public key, 
  - **No**: factory reset,
  - **Cancel**: leave the device alone.

> Would you prefer to keep the discussion on #3?

This issue is not 009a specific, so probably worth keeping it separately.

themrdjj, 2020-10-13:
Was this fixed with d2944ba76221256cc94f10e646fc11898d5cd9ab?

RaghavRao, 2020-10-16:
I believe so, I've been dual booting after updating the SID. Closing.
Great work!!

closed 2020-10-16 by RaghavRao