uunicorn/python-validity #16: Match python-validity egg version to ppa?

RaghavRao, 2020-07-15:
Pip reports python-validity is 0.1 where as the ppa is at 0.7
Should they be equal or is it better to leave them separate?

uunicorn, 2020-07-17:
I'm quite new to Python and Debian packaging, so I don't really know what I am doing. My reasoning is like this: I don't think there is a lot of value in publishing python package alone, since a lot of stuff is coming from the deb maintainer scripts. I was using `setuptools` just to drop everything correctly in the `dist-packages` before wrapping everything in a deb package. The intention was to make the OS package manager control all the files, including the python modules. Pip is not supposed to be used to modify dist-packages, so from that point of view I'm not sure what is the meaning of the python package version. Please let me know if I'm wrong.

RaghavRao, 2020-07-17:
To start, I'm very new to this stuff too, so hopefully someone more knowledgeable can correct us, but the way I see it, the python package is the actual thing you are providing (especially for non debian users since the ppa is not relevant to their machine), but the ppa is the wrapper for the python package.

So if you are cloning the git version and then the ppa catches up, and you want to check if it's different,  you can go by the version number. For example, you can do `apt install python3-systemd` to get the OS supported version, but you can also do `pip install python-systemd` to get the latest pip version. And the only way to tell them apart is to see the pip version.

I believe from this [article](https://stackoverflow.com/questions/15724093/difference-between-python-setup-py-install-and-pip-install) pip calls setup.py anyway, except with added benefits of automatically resolving dependencies, enabling uninstall etc, so it seems like it is always benefecial to do `pip install .` instead of `python ./setup.py install` because pip install will always be a superset.

uunicorn, 2020-07-19:
Thanks, @RaghavRao , that makes a lot of sense. I'll try to keep the versions in sync with the next release. Although I'm not sure you can upgrade a version installed by apt using pip. AFAIK, pip will refuse to uninstall the system version,  so you would probably need to `apt-get remove` the system package first, which means that all the systemd and udev configuration files installed by deb maintainer scripts are gonna disappear.

RaghavRao, 2020-07-24:
You're absolutely right about that. I'm closing this issue as it seems they're in line now. Thank you

closed 2020-07-24 by RaghavRao