uunicorn/python-validity #19: 009a enroll() fails, complains about incorrect arguments

RaghavRao, 2020-07-24:
I'm on the latest commit, fprintd-enroll fails with:

```
Using device /net/reactivated/Fprint/Device/1
Enrolling right-index-finger finger.
EnrollStart failed: Traceback (most recent call last):
  File "/usr/lib64/python3.8/site-packages/dbus/service.py", line 711, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python-validity/dbus-service", line 146, in EnrollStart
    sensor.enroll(uid2identity(uid), 0xf5, update_cb, complete_cb) # TODO parse the finger name
TypeError: enroll() takes 4 positional arguments but 5 were given

```

I ran a factory reset and paired again but no luck.
Any ideas, @uunicorn ?

uunicorn, 2020-07-25:
>  File "/usr/lib/python-validity/dbus-service", line 146, in EnrollStart
>    sensor.enroll(uid2identity(uid), 0xf5, update_cb, complete_cb) # TODO parse the finger name

Looks odd. Line numbers in the stacktrace don't match the latest commit. Also, number of arguments:
https://github.com/uunicorn/python-validity/blob/master/dbus_service/dbus-service#L133

RaghavRao, 2020-07-30:
Thanks, something went wrong in manual installation. I manually copied the files from the repo and it worked

closed 2020-07-30 by RaghavRao