uunicorn/python-validity #21: Exception: Failed: 04c3 after second enroll (on 06cb:009a)

Mstrodl, 2020-08-03:
So I'm not sure if I'm doing something wrong or something, but when I run the DBus service, I experience the following after a firmware reset (following instructions on README):
- First, I can run fprint-enroll without issue
- If I run fprint-verify, my finger doesn't recognize
- If I run fprint-enroll again with another finger (or the same), the service errors with `Exception: Failed: 04c3` and the enroll fails.

Interestingly, there are no issues with using the python API manually, I can both identify and register fingerprints as expected. I can provide more logs if necessary, but obviously accidentally showing my fingerprint data isn't ideal.

uunicorn, 2020-08-04:
> If I run fprint-enroll again with another finger (or the same), the service errors with Exception: Failed: 04c3 and the enroll fails.

This error code means "duplicate record". Right now DBus service uses a [hardcoded id 0xf5]( https://github.com/uunicorn/python-validity/blob/master/dbus_service/dbus-service#L140) each time you attempt to enroll a finger, regardless of what you've passed as a parameter to the `fprintd-enroll`. It means that the driver does not support more than one finger per user right now. Feel free to fix this TODO and raise a PR  :grin: 

However, it does not explain the other problem: why `fprint-verify` does not work for you while identify() does. This is strange. Logs would help.  AFAIK, logs generated from verify/identify should not contain any biometric/sensitive info.

Mstrodl, 2020-08-06:
@uunicorn Actually, I've just tried this again to try and get some debug logs (I've been busy with some personal projects the past week or so) and it seems to be working now, so I guess I'll close this. I'll try and look into what needs to be done with the finger id though!

closed 2020-08-06 by Mstrodl