uunicorn/python-validity #27: Not Working with 06cb:00be

candyman2903, 2020-08-13:
Hello,

I always get the message:
No matching devices found

Please add Support for:
Bus 003 Device 003: ID 06cb:00be Synaptics, Inc.

lsusb -s 003:003 -v

Bus 003 Device 003: ID 06cb:00be Synaptics, Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        16 
  bDeviceProtocol       255 
  bMaxPacketSize0         8
  idVendor           0x06cb Synaptics, Inc.
  idProduct          0x00be 
  bcdDevice            0.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4

Thank you

uunicorn, 2020-08-14:
Hi @candyman2903 , 

`06cb:00be` is not currently supported. 

You can try setting up [synaWudfBioUsb-sandbox](https://github.com/uunicorn/synaWudfBioUsb-sandbox) environment and capturing the pair/enroll/identify logs - this may help to add the support for your device.

dodancs, 2020-11-08:
Hi @uunicorn,

I have tried to extract the dlls which come for this fingerprint reader, but did not get any `*.xpfwext` file. Is this an issue? This is the directory listing I got after extracting the driver:
```
'../code$GetExtractPath$/':
./  ../  Source/  setup.cmd

'../code$GetExtractPath$/Source':
./  ../  Goodix/  Synaptics/

'../code$GetExtractPath$/Source/Goodix':
./  ../  Fp_data_processor.signed.dll  GoodixEngineAdapter.dll  sgx_white_list_cert.bin  usbinterface.cat  usbinterface.dll  usbinterface.inf  WBDI_Data_Enclave.signed.dll

'../code$GetExtractPath$/Source/Synaptics':
./  ../  sgx_capable.dll  synaFpAdapter107.dll  synaTEE107.signed.dll  synaumdfsgx.cat  synaWudfBioUsbSGX107.dll  synaWudfBioUsbUwpSgx.inf
```

Then I tried to launch the `run.sh` program, which resulted in [https://pastebin.com/0T5C1g5R](https://pastebin.com/0T5C1g5R)

uunicorn, 2020-11-09:
Hi @dodancs ,

Not having a `*.xpfwext` seems to be normal for some sensors. However you do need a driver DLL. Looking at your file list `synaWudfBioUsbSGX107.dll` appears to be the closest match. However I really doubt Wine approach is going to ever work with SGX enclaves.

Regards,
Viktor

dodancs, 2020-11-09:
> Hi @dodancs ,
> 
> Not having a `*.xpfwext` seems to be normal for some sensors. However you do need a driver DLL. Looking at your file list `synaWudfBioUsbSGX107.dll` appears to be the closest match. However I really doubt Wine approach is going to ever work with SGX enclaves.
> 
> Regards,
> Viktor

Do you have suggestions on what to do in this case to get what you need to implement support also for this sensor?

uunicorn, 2020-11-09:
Nah, sorry. I have a feeling that dealing with SGX could take way more time than I'm prepared to spend on it. Are you sure there are no non-SGX versions of this driver?

dodancs, 2020-11-09:
> Nah, sorry. I have a feeling that dealing with SGX could take way more time than I'm prepared to spend on it. Are you sure there are no non-SGX versions of this driver?

What does the non-SGX version mean? I am not sure what to look for.

Thanks for the engagement! I appreciate your work and the time you put in very much.

himekifee, 2020-11-09:
[SGX](https://en.wikipedia.org/wiki/Software_Guard_Extensions) basically makes reverse engineering harder. From your list of file, looks like all critical codes are protected by SGX since they have SGX in their file name. You can try to find older version of that sensor driver. A none SGX protected synaWudfBioUsb.dll is what to investigate.

memiks, 2021-02-03:
Hi all,

I found a non SGX driver on my Yoga 950i and execute your commands.
but the wine docker image lack of two dlls:
WppRecorderUM.dll
API-MS-Win-devices-config-L1-1-2.dll

and you can find last log in attachments... I think this is not useful
[1612320909.log](https://github.com/uunicorn/python-validity/files/5915782/1612320909.log)
...

Please tell me how I could help...

himekifee, 2021-02-03:
You can join the [discord](https://discord.gg/tJzmxpeF) server for quicker help.

AnterCreeper, 2022-03-02:
Non-SGX version of my fingerprint driver(Should be?)
Dll file doesn't have export function of DLLGetClassObject at all........

running the SGX version of the driver, the wine crashed.

AnterCreeper, 2022-03-02:
my device is 06cb 00d2,
which should be specially for HewlettPackard.

AnterCreeper, 2022-03-02:
What i found is
the protocal is same for:
USB\VID_06CB&PID_00BB
USB\VID_06CB&PID_00D2
USB\VID_06CB&PID_00F4

srijanmukherjee, 2024-01-06:
Any updates on this?