uunicorn/python-validity #31: usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

miguelmota, 2020-08-19:
Steps to reproduce:

```bash
$ cd /usr/share/python-validity/playground
$ python3
Python 3.8.5 (default, Jul 27 2020, 08:42:51)
[GCC 10.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from prototype import *
>>> open9x()
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 223, in get_interface_and_endpoint
    return self._ep_info[endpoint_address]
KeyError: 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mota/.local/lib/python3.8/site-packages/validitysensor/init.py", line 48, in open
    open_common()
  File "/home/mota/.local/lib/python3.8/site-packages/validitysensor/init.py", line 29, in open_common
    init_flash()
  File "/home/mota/.local/lib/python3.8/site-packages/validitysensor/init_flash.py", line 113, in init_flash
    info = get_flash_info()
  File "/home/mota/.local/lib/python3.8/site-packages/validitysensor/flash.py", line 40, in get_flash_info
    rsp = tls.cmd(unhex('3e'))
  File "/home/mota/.local/lib/python3.8/site-packages/validitysensor/tls.py", line 124, in cmd
    rsp = self.usb.cmd(cmd)
  File "/home/mota/.local/lib/python3.8/site-packages/validitysensor/usb.py", line 102, in cmd
    self.dev.write(1, out)
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 940, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 215, in setup_request
    intf, ep = self.get_interface_and_endpoint(device, endpoint_address)
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 225, in get_interface_and_endpoint
    for intf in self.get_active_configuration(device):
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 236, in get_active_configuration
    self.managed_open()
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.8/site-packages/usb/core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/usr/lib/python3.8/site-packages/usb/backend/libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "/usr/lib/python3.8/site-packages/usb/backend/libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/usr/lib/python3.8/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
>>>
```

Additional information

Fingerprint succesfully enrolled

```bash
$ fprintd-list mota
found 1 devices
Device at /net/reactivated/Fprint/Device/2
Using device /net/reactivated/Fprint/Device/2
Fingerprints for user mota on DBus driver (press):
 - #0: WINBIO_FINGER_UNSPECIFIED_POS_01
```

Device `06cb:009a Synaptics, Inc.`

```bash
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 015: ID 06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader
Bus 001 Device 004: ID 04f2:b604 Chicony Electronics Co., Ltd Integrated Camera (1280x720@30)
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 002: ID 1050:0406 Yubico.com Yubikey 4/5 U2F+CCID
Bus 001 Device 006: ID 2386:432f Raydium Corporation Touch System
Bus 001 Device 009: ID 046d:c069 Logitech, Inc. M-U0007 [Corded Mouse M500]
Bus 001 Device 008: ID 2516:001a Cooler Master Co., Ltd. Storm Quick Fire XT
Bus 001 Device 010: ID 2972:0047 FiiO Electronics Technology USB 2.0 Hub
Bus 001 Device 007: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

udev rules

```bash
$ cat /etc/udev/rules.d/30-fpreader.rules
ATTRS={idVendor}=="06cb", ATTRS{idProduct}=="009a", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="06cb", ATTRS{idProduct}=="009a", ATTRS{dev}=="*", TEST=="power/control", ATTR{power/control}="auto"
SUBSYSTEM=="usb", ATTRS{idVendor}=="06cb", ATTRS{idProduct}=="009a", ENV{LIBFPRINT_DRIVER}="Synaptics Sensors"
```

OS and kernel version

```bash
$ uname -a
Linux archlinux 5.8.1-arch1-1 #1 SMP PREEMPT Wed, 12 Aug 2020 18:50:43 +0000 x86_64 GNU/Linux
```

uunicorn, 2020-08-19:
Hi @miguelmota ,

To use the scripts under `playground` directory you must be either root or change the ownership of the device file under `/dev/bus/usb/`. I used to have an udev rule which would set the ownership for me:
```
$ cat /etc/udev/rules.d/10-fpreader.rules
ATTRS{idVendor}=="138a", ATTRS{idProduct}=="0097", OWNER="unicorn"
```

Also, if you choose to run playground scripts as a non-root user, you may need to check permissions for `/sys/class/dmi/id/product_name` and `/sys/class/dmi/id/product_serial` as well. Those files are used to obfuscate the host's TLS key before storing storing it on the device. If they are not readable by the user, it falls back to hardcoded `VirtualBox` and `0` values. Which means that you may need to perform a factory reset each time you switch between playground scripts (running as a non-root user) and the DBus service (running as root).

Hope this helps.

miguelmota, 2020-08-19:
Thanks! `product_name` was readable but not `product_serial` so this fixed it:

```bash
$ sudo chmod +r /sys/class/dmi/id/product_serial

$ ls -l /sys/class/dmi/id/product_serial
-r--r--r-- 1 root root 4096 Aug 19 16:21 /sys/class/dmi/id/product_serial
```

closed 2020-08-19 by miguelmota

miguelmota, 2020-08-19:
FYI the `/sys/class/dmi/id/product_serial` permissions get overridden on reboot so it requires running `chmod` command again. Not sure if there's a way to persist it.