uunicorn/python-validity #32: 06cb:009a - login keyring prompt on first boot

faaizajaz, 2020-08-25:
Hi and thanks for all the amazing work on this project.

I am running this with the 06cb:009a on Ubuntu 20.04 with Gnome 3.34, and while this works as expected, on first login (after a shutdown) I am able to get past the login screen using the fingerprint, but once on the desktop I am prompted to enter my password with the message:
```The login keyring did not get unlocked when you logged into your computer```.

I suspect this is a Gnome issue (presumably it does not let fingerprint unlock the keyring and requires the password) but I thought I'd add it here just in case.

uunicorn, 2020-08-26:
Hi @faaizajaz ,

I think it is the intended behavior if you wish your `login` keyring to be protected/encrypted.

The fingerprint authentication itself yields no secrets which could be used for en(de)cryption, so a secret (keyring encryption key) must be stored elsewhere. Storing it on the disk is equivalent to leaving your keyring unencrypted. Another option is to store the secret on the sensor device itself and associate it with your finger. Technically it is possible. Moreover `python-validity` already supports associating arbitrary blobs with a finger record. However in this case your keyring contents is going to be as secure as the template database on the sensor. And the way the existing Validity crypto works, anyone with a physical access to your computer can extract the contents of your template database. Including any secrets associated with enrolled fingers. There are ways to slightly improve this situation, but then again - how secure the fingerprint auth should really be? You usually leave your latent "key material" all over the device which you're trying to protect, so it can't be very secure in principle.

wistarine, 2021-01-03:
> Hi and thanks for all the amazing work on this project.
> 
> I am running this with the 06cb:009a on Ubuntu 20.04 with Gnome 3.36.3, and while this works as expected, on first login (after a shutdown) I am able to get past the login screen using the fingerprint, but once on the desktop I am prompted to enter my password with the message:
> `The login keyring did not get unlocked when you logged into your computer`.
> 
> I suspect this is a Gnome issue (presumably it does not let fingerprint unlock the keyring and requires the password) but I thought I'd add it here just in case.

same as me ,how do you u fix this keyring sitiuation?

Fuseteam, 2021-09-13:
> Another option is to store the secret on the sensor device itself and associate it with your finger. Technically it is possible. Moreover python-validity already supports associating arbitrary blobs with a finger record. 

this sounds like a reasonable option tbh
especially considering

> You usually leave your latent "key material" all over the device which you're trying to protect, so it can't be very secure in principle.

and i mean usually when someone gets physical access to the hardware security is pretty much lost anyway

Anifyuli, 2023-05-10:
I get same problem too. Can someone give me simple solutions or recommendation commands to auto sign that keyring?

Anifyuli, 2023-05-10:
I try to disable GNOME keyring from updating PAM settings and this issue not appear again
[image]

I feel confuse what I do is safe or not. Sorry My English

Fuseteam, 2023-07-26:
@Anifyuli a simple workaround is by setting a blank password. However this is by no means secure

Anifyuli, 2023-07-26:
@Fuseteam Oh, I see. But I get trouble again in Ubuntu 23.04. `python3-validity` always killed if resume after suspend