uunicorn/python-validity #34: possible to expand this to Kensington Validity 06cb:0088 ?

jontis, 2020-08-29:
Hi, I started following your:
https://github.com/uunicorn/synaWudfBioUsb-sandbox

But encountered a slightly different folder structure and that's about where my skills end.
The item is a Kensington Verimark USB fingerprint reader.
If you could determine if it is suitable for this project and guide me how to proceed, I'd be happy to help.

Folder structure of the driver is:
KensingtonFingerprintApplication.exe
KensingtonFingerprintApplication.lnk
synaBscAdapter66.dll
SynaCP66.dll
SynaFPCoInstaller.dll
SynaFPService.exe
synahidumdf.sys
SynaU2F.exe
synaumdf.cat
synaWudfBioUsb66.dll
synaWudfBioUsbKensProd.inf
synaWudfU2fhid.dll
synaWudfU2fhid.ini
WudfUpdate_01011.dll

[synaWudfBioUsbKensProd.inf.txt](https://github.com/uunicorn/python-validity/files/5146037/synaWudfBioUsbKensProd.inf.txt)

uunicorn, 2020-08-30:
Looking at the .inf file, it appears that the driver GUID is the same as for `0097` and `009a` (`96710705-B080-4B29-A3EC-B16935AE663A`), which means that [synaWudfBioUsb-sandbox](https://github.com/uunicorn/synaWudfBioUsb-sandbox) approach might just work. Have you tried renaming `synaWudfBioUsb66.dll` to `synaWudfBioUsb.dll` and following the rest of `synaWudfBioUsb-sandbox` instructions? I know, it's not an easy task. I've tried to make `synaWudfBioUsb-sandbox` and `wine` setup to be a bit more user friendly but got bogged down by the Wine packaging and gave up.

Do you know if your sensor is Match-on-Sensor or Match-on-Host?

oofnikj, 2020-09-03:
@uunicorn I'm about done writing up some instructions for building the patched Wine in a Debian container, happy to open a PR for it if you're interested.

uunicorn, 2020-09-05:
> There is no such file.

Well, fwext is just an overlay on top of firmware in ROM. Perhaps it is not needed for this device.
The .inf file has an empty `[FWextCopy]` section, which implies that it is expected.
`synaWudfBioUsb-sandbox` may still work.

> The sensor is claimed to be match-on-host.

This is unfortunate. `python-validity` lacks support for MoH devices at the moment.
I've provided a bit more details in [this comment](https://github.com/uunicorn/python-validity/pull/37#issuecomment-687098106).

jontis, 2020-09-05:
Interesting. Thank you for explaining. 

In that case, I might get it's younger sibling to experiment with adding support. 
It announced to be match-in-sensor.

Kensington VeriMark IT Fingerprint Key

What do you think of this?

uunicorn, 2020-09-05:
> What do you think of this?

I think we can definitely try. Probably worth raising another issue with a correct usb id.

jaxjexjox, 2021-01-12:
I apologise, I think I've gone and made a new thread, when I should have read this one.
Am I correct in interpreting this discussion as the :0088 device isn't going to work / may not ever work?

himekifee, 2021-01-12:
> I apologise, I think I've gone and made a new thread, when I should have read this one.
> Am I correct in interpreting this discussion as the :0088 device isn't going to work / may not ever work?

You should look for [Validity90](https://github.com/nmikhailov/Validity90), that one is generally for MoH.

uunicorn, 2021-01-12:
@jaxjexjox 

> isn't going to work / may not ever work?

It is not that it may never work, it is just (unlike `Validity90`) there is no existing framework to do the fingerprint matching inside the `python-validity` code at the moment. You are more than welcome to contribute it though.

jaxjexjox, 2021-01-13:
> You should look for [Validity90](https://github.com/nmikhailov/Validity90), that one is generally for MoH.

I thought it was all but abandoned several years ago?

himekifee, 2021-01-13:
Well, yes, but that is the most relative project.

uunicorn, 2021-01-13:
I think @3v1n0 's branch of `fprintd` is still alive: https://github.com/3v1n0/libfprint

jaxjexjox, 2021-01-13:
Thanks for your hard work and replies, people like you make using free software awesome, appreciate it.

jontis, 2021-01-13:
About Validity90, I just want to add that I wouldn't call it abandoned. I think the project was making progress and had a lot of traction among users. Then the owner disappeared. No idea if something happened to him, or if he got tired of the project and made a new github identity. I believe something happened to him, or he would have transferred ownership of the project. 3v1n0 picked it up, but it doesn't seem to keep the same pace.

parkerlreed, 2021-04-26:
Found another one in the wild

ID 06cb:0089 Synaptics, Inc. 

https://www.amazon.com/gp/product/B07D3N3QHJ/

Works with Windows Hello after it grabs a Synaptics driver. 

```
Bus 001 Device 005: ID 06cb:0089 Synaptics, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        16 
  bDeviceProtocol       255 
  bMaxPacketSize0         8
  idVendor           0x06cb Synaptics, Inc.
  idProduct          0x0089 
  bcdDevice            1.54
  iManufacturer           0 
  iProduct                0 
  iSerial                 1 b77c94008419
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              10
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```