uunicorn/python-validity #35: Is there a way to disable fringerprint when Laptop is docking( lid Closed)

plumlis, 2020-08-31:
Is there a way to disable fringerprint when Laptop is docking( lid Closed)
Somethimes I use my laptop with a thunderbolt3 dock. and ld is closed.
It still tell  me I have to use fingerprint to get sudo

plumlis, 2020-08-31:
Or only use fingerprint to unlock.

uunicorn, 2020-08-31:
Yeah, it is sort of `sudo` and/or PAM module problem, not the `python-validity` itself. In theory it should be possible to authenticate with either a finger or a password *in any order*, in practice however it is all down to the actual app which uses PAM API. While Gnome screen lock/login seems to be smart enough to run 2 separate PAM stacks at the same time, `sudo` is not. 

I think there are ways around, but we inherited this behavior from the upstream [fprintd PAM module](https://gitlab.freedesktop.org/libfprint/fprintd/-/blob/master/pam/README#L24):
```
* pam_fprintd doesn't support entering either the password or a fingerprint,
  as pam_thinkfinger does, because it's a gross hack, and could be fixed
  by having the login managers run 2 separate PAM stacks
```

This is how I understand it anyway. I'm not very familiar with that part of the stack.

plumlis, 2020-09-22:
I tried to add 

`auth            sufficient      pam_unix.so try_first_pass likeauth nullok`

in file 

`sudo vim /etc/pam.d/sudo`

and it seems work but not perfect.

I saw howdy https://github.com/boltgolt/howdy has a option in conf to disable face unlock while docking. Maybe fingerprint can be done like that.

kakawait, 2020-09-22:
I'm not dev nor affiliated to `python-validity` but I'm curious about subject. When checking _Howdy_ we can see that is using this own pam _module_ (I'm not sure about term _module_).

https://github.com/boltgolt/howdy/blob/master/src/pam.py#L30

I think you can create your own with same logic.

I don't think it should be part of `python-validity` because `python-validity` is way more low level stack.