uunicorn/python-validity #40: Support for 06cb:00c7

giovannicaligaris, 2020-09-13:
Has anyone been able to get 06cb:00c7 to work?

I am happy to help if someone can guide me on how to do it.

uunicorn, 2020-09-14:
Do you know if it is a Match-on-Chip or Match-on-Host device?

giovannicaligaris, 2020-09-14:
@uunicorn Sorry, I am a bit confused. What do you mean by that?

uunicorn, 2020-09-14:
@giovannicaligaris some sensors are capable of identifying a finger without exposing any biometric information to the host computer. Those are Match-on-Chip. Can you share a link to a Windows driver for your sensor?

giovannicaligaris, 2020-09-14:
@uunicorn You can download the HID drivers from here:

https://novacustom.stackstorage.com/s/s2zVcQyd004hRQJ?dir=%2FPC50D-PC70D%2FDrivers&node-id=176677

The model of the laptop is Clevo PC50DD2

uunicorn, 2020-09-14:
Your driver seems to be [here](
https://novacustom.stackstorage.com/s/s2zVcQyd004hRQJ?dir=%2FPC50D-PC70D%2FOptions%2F02_Finger%2FSynaptics_Gamma&node-id=179613). It is not quite clear from the `.inf` file if it is MoH or MoC.

If you manage to set up the [synaWudfBioUsb](https://github.com/uunicorn/synaWudfBioUsb-sandbox) project and gather some logs it could be possible to add support for this sensor.

Matombo, 2020-12-14:
Hi,
I try to setup the sandbox, but the driver does not seem to contain the *.xpfwext file?

himekifee, 2020-12-14:
> Hi,
> I try to setup the sandbox, but the driver does not seem to contain the *.xpfwext file?

Can you capture some packets? Some drivers are pretty easy to reverse engineer. If it doesn't have a firmware file, it might be in that case.

Matombo, 2020-12-23:
Sry took some time till i could look into this again, and tomorrow i'm on holiday so my next answer will be in the new year

I managed to setup the sandbox, i removed the lines for the xpfwext file from the run script an altered the device command as the device is called just "Synaptic" and not "Synaptic Fingerprint" but sadly the driver crashes. log is attached

[1608738488.log](https://github.com/uunicorn/python-validity/files/5736004/1608738488.log)

Matombo, 2021-01-12:
Installed a windows test partition with wireshark on the device.
Now i can sniff usb packets. What am I looking for exactly?/What iformation do you need to support the device?

I found one packet with 35kb, so I guess it's a match on host device?
However throwing both file and binwalk at the extracted payload did yiel no image format whatsoever, it did yield nothing at all to be precise. I read that some synaptic sensors encrypt the image. Could this be the case?

uunicorn, 2021-01-12:
Almost all the USB communication between the sensor and the host is TLS encrypted. There are two ways to work around it.

1. Use Windows debugger to extract the session keys from the running driver
2. Use a fake Windows environment which logs all encryption operations so that you can see the cleartext for a the comms.

I think the original Validity90 instructions go with the first approach. If you manage to extract the session keys, you can decrypt the traffic which you've captured with Wireshark. However it is not easy. I think original instructions are specific to the driver version and won't work as-is on any other driver/device without some extensive reverse engineering.

With `synaWudfBioUsb-sandbox` I've used the second approach. I've hacked Wine sources to make them intentionally insecure and log all the crypto operations. It is a more generic approach, however it still assumes a lot about the nature of the driver and it could be challenging to get it working.

I've looked at the logs from your previous comment (`1608738488.log`) it appears that the `a.exe` can't load the driver's dll for some reason. Can you list the contents of your folder where you were running it and the contents of the script?

> I found one packet with 35kb, so I guess it's a match on host device?

Not necessary, the match-on-chip devices exchange some chunky blobs which contain no biometric data.

Matombo, 2021-01-13:
The folder got deleted by accident so I cloned and ran the sandbox again
```
$ ls -la
insgesamt 3064
drwxrwxr-x  3 test test    4096 Jan 13 10:40 .
drwxr-xr-x 17 test test    4096 Jan 13 10:36 ..
-rw-rw-r--  1 test test   27550 Jan 13 10:40 1610530840.log
-rwxrwxr-x  1 test test 1159527 Jan 13 10:23 a.exe
-rw-rw-r--  1 test test    3023 Jan 13 10:23 breakpoints.c
-rw-rw-r--  1 test test     252 Jan 13 10:23 breakpoints.h
-rw-rw-r--  1 test test      98 Jan 13 10:23 .env
drwxrwxr-x  8 test test    4096 Jan 13 10:23 .git
-rw-rw-r--  1 test test      57 Jan 13 10:23 .gitignore
-rw-rw-r--  1 test test   82966 Jan 13 10:23 hello.cc
-rw-rw-r--  1 test test     265 Jan 13 10:23 mk
-rw-rw-r--  1 test test   56576 Jan 13 10:23 notes.txt
-rw-rw-r--  1 test test    1868 Jan 13 10:23 README.md
-rwxrwxr-x  1 test test     305 Jan 13 10:26 run.sh
-rw-rw-rw-  1 test test   18440 Apr 17  2020 SetupInf.exe
-rw-rw-rw-  1 test test      40 Apr 17  2020 SetupInf.ini
-rw-rw-rw-  1 test test  186440 Apr 17  2020 synaFpAdapter115.dll
-rw-rw-rw-  1 test test   12694 Apr 17  2020 synaumdf.cat
-rw-rw-rw-  1 test test 1517136 Apr 17  2020 synaWudfBioUsb115.dll
-rw-rw-rw-  1 test test    5959 Apr 17  2020 synaWudfBioUsbUwp.inf
-rw-rw-r--  1 test test      10 Jan 13 10:17 usb.txt
```

I altered the run script to ignore the missing xpfwext file and hard coded the device path because the awk command didn't work:
```
$ cat run.sh 
#!/bin/sh
set -e

DEVICE="/dev/bus/usb/001/003"
PWD=$(pwd)
BASENAME=$(basename "$PWD")

docker run --rm -it \
    --device $DEVICE \
    --env-file .env \
    -v $PWD:/$BASENAME \
    -v $PWD/usb.txt:/root/.wine/drive_c/usb.txt \
    wine:validity wine64 "/${BASENAME}/a.exe" "$@" | tee "$(date +%s)".log
```

uunicorn, 2021-01-13:
Try renaming `synaWudfBioUsb115.dll` to `synaWudfBioUsb.dll` and running the script again. `a.exe` looks for `synaWudfBioUsb.dll` file.

Matombo, 2021-01-13:
Got further now some windows dlls are missing

[1610531337.log](https://github.com/uunicorn/python-validity/files/5807744/1610531337.log)

himekifee, 2021-01-13:
You may download them from the Internet.

Matombo, 2021-01-13:
got WppRecorderUM.dll from the windows test partition and api-ms-win-devices-config-l1-1-2.dll from https://www.dll-files.com/api-ms-win-devices-config-l1-1-1.dll.html because i couldn't find it on my windows install?

However it now finds WppRecorderUM.dll but  /api-ms-win-devices-config-l1-1-1.dll is still not loaded

Matombo, 2021-01-13:
i tried making the filename all lowercase

himekifee, 2021-01-13:
import_dll Library `api-ms-win-devices-config-l1-1-2.dll` l-1-1-"2"

Matombo, 2021-01-13:
ah the library exists in a 32-bit and 64-bit version thats why it didn't get loaded
and the find command ist case sensitive, that's why i didn't found it on my windows partition

Matombo, 2021-01-13:
another step further: DllGetClassObject was not exported from the synawudfbiousb.dll

[1610532950.log](https://github.com/uunicorn/python-validity/files/5807908/1610532950.log)

uunicorn, 2021-01-13:
That's odd. Can you share a link to the driver which you are using?

himekifee, 2021-01-13:
Uhh, new structure

Matombo, 2021-01-13:
> That's odd. Can you share a link to the driver which you are using?

the one you referenced here: https://github.com/uunicorn/python-validity/issues/40#issuecomment-692373686

but i found another version on the clevo website, will test it now

Matombo, 2021-01-13:
Ok the driver from the clevo website yielded the same result but the nongamma driver from here https://novacustom.stackstorage.com/s/s2zVcQyd004hRQJ?dir=PC50D-PC70D%2FOptions%2F02_Finger&node-id=0 got a result
[1610533383.log](https://github.com/uunicorn/python-validity/files/5807967/1610533383.log)

uunicorn, 2021-01-13:
This looks more promising. Can you check that the device path is calculated correctly in `run.sh`?
```
DEVICE=$(lsusb | grep 'Fingerprint' | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $2 "/" $3 }')
```

himekifee, 2021-01-13:
Actually do you want to have a discord server @uunicorn ? Issue seems not to be a good place to do experiments.

Matombo, 2021-01-13:
Yes I have:

Bus 001 Device 004: ID 06cb:00c7 Synaptics, Inc.

But I already recognized it and hard coded the path in run.sh:

DEVICE="/dev/bus/usb/001/003"

himekifee, 2021-01-13:
You may inspect your docker environment first.

Matombo, 2021-01-13:
Ok it seems to interact now with the device:

[1610534760.log](https://github.com/uunicorn/python-validity/files/5808093/1610534760.log)

it hangs at "All done, sleeping" till i press ctrl + c

after that i can't rerun the script without rebooting first

uunicorn, 2021-01-13:
Ok, I think there are a few problems. 
I guess you'll need to get the `DEVICE` variable to work properly first. As part of initial pairing the sensor is rebooted several times. When this happens the host detects USB reconnect and allocates a new device number, so your hardcoded `/dev/bus/usb` path no longer works. Whenever device is rebooted so is the Windows driver. Windows driver expects to be "activated" by the framework, but instead of framework we have this `a.exe`, so you would probably need to run it a few times. Each time the logs may look differently.

Another unrelated problem: according to the logs, flash partition command resulted in an error. Not sure why yet.

uunicorn, 2021-01-13:
@himekifee , I've created a `python-validity` Discord server, but I'm not very familiar with Discord.

himekifee, 2021-01-13:
You may need to post a invite link instead of the name. The name can be duplicated for multiple servers.

uunicorn, 2021-01-13:
https://discord.gg/f3DdJtJ4

Matombo, 2021-01-13:
> I guess you'll need to get the `DEVICE` variable to work properly first.

I'm on it

> Another unrelated problem: according to the logs, flash partition command resulted in an error. Not sure why yet.

I deleted this line from run.sh `-v $PWD/$XPFWEXT:/root/.wine/drive_c/system32/$XPFWEXT \`

Does it have to do something with it?

Matombo, 2021-01-13:
discussion continues on the discord server: #general

Matombo, 2021-01-15:
A short summary of discord:

- The Gamma driver uses UMDF v2: https://github.com/uunicorn/synaWudfBioUsb-sandbox was written for UMDF v1 and needs a complete rewrite to reverse the driver.
- From strings extracted from the driver binary: the 00c7 is propably Match on Host: Bad news as the open source matching algorithm used by libfprint is quite lacking for non full fingerprints. That's why currently only match on device synaptics devices are supported
- The non gamma driver does not support 00c7, see below

I think I found out what the gamma and non Gamma stands for:
https://www.synaptics.com/company/news/clevo-securepad-gamma
tl;dr: The 00c7 is a glass touchpad and I think exactly the model the press article is refering to. So it only works with the gamma driver. I guess the device the driverpage above is for was sold with an older touchpad/fp combo too, so that's why the older driver is also in there.

mebenstein, 2021-04-14:
Any progress on this? I would really like to use this model on Arch, if there's anything I can help with let me know.

Matombo, 2021-05-12:
Sadly it a match on host device: So it is blocked by no available open source partial fingerprint matching algorithm: https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/272 https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/271

tintin10q, 2025-07-08:
I want to add these links to windwos drivers:

https://skikk.stackstorage.com/s/wVqaxQukZKlmtaf/en/files/647002