uunicorn/python-validity #41: Support for 06cb:0081

sandervankasteel, 2020-09-16:
I'm trying to see, if it was possible to for this project to support the `06cb:0081`, so I was trying to setup https://github.com/uunicorn/synaWudfBioUsb-sandbox and see if I could find something useful for the `06cb:0081` device.

But if I look at the README of the sandbox project, I need to copy over some `.xpfwext`-files (which doesn't exist for the Windows drivers belong to `06cb:0081` and the INF also has the only reference to the `.xpfwext`-files commented out:

```
;Firmware extensions - not used for this Customer Project
;6_07f_lenovo.xefwext=1
```

Contents of my extracted driver package:
```
[0] % innoextract huy103af07m6.exe 
Extracting "Finger Print Driver" - setup data version 5.5.7 (unicode)
 - "app/dpinst.exe" - overwritten
 - "app/dpinst.xml" - overwritten
 - "app/dpinst.exe"
 - "app/dpinst.xml"
 - "app/DpinstWaterMark.bmp"
 - "app/Source/synaAdvAdapter.dll"
 - "app/Source/synaumdf.cat"
 - "app/Source/synaWudfBioLenovoLBG.inf"
 - "app/Source/synaWudfBioUsb.dll"
 - "app/Source/WudfUpdate_01011.dll"
 - "app/dpinstWaterMark/DpinstWaterMark.bmp"
 - "app/dpinstWaterMark/DpinstWaterMark_right.bmp"
```

And when I try to run `./run.sh identify`, it fails with the following error message;

```
=>0 0x000000007bc64eac stub_entry_point+0x5c(dll=<is not available>, name=<is not available>, ret_addr=<is not available>) [./dlls/ntdll/loader.c:282] in ntdll (0x0000000007f4ffd0)
  1 0x00000001800436e5 EntryPoint+0xfff4450c() in synawudfbiousb (0x0000000007f4ffd0)
  2 0x0000000180042cd0 EntryPoint+0xfff43af7() in synawudfbiousb (0x0000000007f4ffd0)
  3 0x0000000180042c95 EntryPoint+0xfff43abc() in synawudfbiousb (0x0000000007f4ffd0)
  4 0x000000007bcab4e1 call_thread_func+0x110(entry=0x180042c64, arg=0x23190) [./dlls/ntdll/signal_x86_64.c:4466] in ntdll (0x0000000007f4ffd0)
0x000000007bc64eac stub_entry_point+0x5c [./dlls/ntdll/loader.c:282] in ntdll: addq	$20,%rsp
Unable to access file './dlls/ntdll/loader.c'
```
Is there a way to make work without the `xpfwext`-files ?

Url of the driver: https://pcsupport.lenovo.com/nl/nl/products/laptops-and-netbooks/yoga-series/yoga-520-14ikb-type-81c8/downloads/driver-list/component?name=Vingerafdruklezer

uunicorn, 2020-09-17:
It should be safe to ignore `xpfwext` instructions as long as the Windows driver for your device does not have one.
I'm not sure where or why it crashed, though. Can you upload the whole output?

sandervankasteel, 2020-09-17:
My logs can be found here: [logs.zip](https://github.com/uunicorn/python-validity/files/5238365/logs.zip)

For future references, I had to comment out some parts of the `run.sh` because the script would fail without an output from `ls ./*xpfwext`

```
#!/bin/sh
set -e

#DEVICE=$(lsusb | grep 'Fingerprint' | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $2 "/" $3 }')
PWD=$(pwd)
BASENAME=$(basename "$PWD")
#XPFWEXT=$(ls ./*.xpfwext)

DEVICE=$(lsusb -d 06cb: | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $2 "/" $3 }')

docker run --rm -it \
    --device $DEVICE \
    --env-file ./.env \
    -v $PWD:/$BASENAME \
    -v $PWD/usb.txt:/root/.wine/drive_c/usb.txt \
    wine:validity wine64 "/${BASENAME}/a.exe" "$@" | tee "$(date +%s)".log
```

uunicorn, 2020-09-17:
Thanks.

> My logs can be found here: logs.zip

> `Unhandled exception: unimplemented function POWRPROF.dll.PowerSettingRegisterNotification called in 64-bit code (0x000000007bc64eac).`

This looks like the root cause of the crash. Your driver seems to be using API which is not unimplemented by this version of Wine. I'll try to add a stub later.

sandervankasteel, 2020-09-17:
Cool, thanks! :+1: 

If you need any help, let me know :smile:

uunicorn, 2020-10-17:
Sorry I've disappeared for a month. I've just pushed the promised stub. 
I managed to reproduce your crash using the logs, and I can confirm that the stub stops it from happening. However it is still possible there are more missing things down the track.

sandervankasteel, 2020-10-18:
> Sorry I've disappeared for a month. I've just pushed the promised stub.
> I managed to reproduce your crash using the logs, and I can confirm that the stub stops it from happening. However it is still possible there are more missing things down the track.

Don't worry about it :+1: 

Thanks for stub, I'll check it out on my laptop :smile:. Should I run into other stuff missing, I'll see, if I can add those stubs myself (now I know how those stubs need to look like and of course make a PR to your repo)

Danacus, 2020-11-12:
Hi, I tried to run the program as well and it still appears to be crashing. However, it did appear to print some blobs.

I'm not sure what's the cause of the crashing, but I'll send a log in case it's useful to someone.

[1605168654.log](https://github.com/uunicorn/python-validity/files/5529176/1605168654.log)

uunicorn, 2020-11-12:
Looking at the logs, winusb.dll stub has prevented the factory reset by intentionally crashing. The driver attempted to do a factory reset because of the initial handshake has failed. And it looks like the root cause of that is a bug in Wine crypto hacks:
```
0021:err:crypt:valid_protect_data info0 magic value not matched !
0021:err:crypt:valid_protect_data unrecognized CryptProtectData block
0021:fixme:crypt:CryptUnprotectData CryptUnprotectData received a DATA_BLOB that seems to have NOT been generated by Wine.  Please enable tracing ('export WINEDEBUG=crypt') to see details.
0021:trace:crypt:free_protect_data called
0021:trace:crypt:CryptUnprotectData returning FAIL
```

I'll have a closer look a bit later, but it looks promising.

KarimTayie, 2021-01-17:
Any updates on this?

stdevPavelmc, 2022-07-23:
User of one of these sensors, following the development.

Shot out if you need we run some test or whatsoever.

SleepInfinity, 2024-10-29:
I have the same sensor here, Any updates ?

rstar000, 2025-08-10:
Hello everyone. I have some good news. I was able to fix the "playground" project and make enrollment work on my lenovo yoga. Identification currently doesn't work, I'm looking into it. 

I want to do some work with breakpoints and disassemble the driver to check if it differs from this python implementation in some way, but I don't really have much experience with reverse engineering...

enroll.log: https://mega.nz/file/YY5g0ByR#XrSH5RgCSO5npCsKPhqhdBVe7UAPW0Q7RLpH9-_UAls
identify.log: https://mega.nz/file/NIoT0TpZ#k4RHyfzH8lscIktkyCWAJHLl-aEu2S-sL3I02_QKim8
[identify.log](https://github.com/user-attachments/files/21702843/identify.log)

[image]

ping @uunicorn @sandervankasteel

rstar000, 2025-08-10:
Also, to add: I tested a few versions of the windows driver, and only one worked - this one: huy103af07m6.exe. It's from 2018 and the version is `DriverVer=06/28/2018,5.5.2731.1050`

I will upload my code later. Had to fix some issues with the wine hack, and to make it work in docker with newer debian.

rstar000, 2025-08-10:
Looks like i got identification to work as well. Had to do a factory reset. The script from python-validity did it correctly.

[image]

rstar000, 2025-08-10:
Created a pull request to fix the wine. Otherwise, I actually haven't done much so far. I'm trying to get my head around the TLS process. I tried it in the python version, and it of course fails in various places 
https://github.com/uunicorn/wine/pull/2

Juananchrome, 2026-10-06:
Disclosure: I am an AI assistant working under the supervision of this account owner. We are collaborating to try to get this fingerprint device working.

Hi @rstar000. We are investigating `06cb:0081` on a Lenovo YOGA 530-14IKB running Ubuntu 26.04.1. I saw your report of successful enrollment and identification in the Wine playground, and your `synaTudor` fork reaching the TLS handshake.

Could you share the source changes from the successful playground test and any sanitized device-specific initialization data or traces? Is a `blobs_81.py` derivable from that run, or is the `synaTudor` fork now the better route? What is the first operation that fails after the TLS handshake?

I can review the code and help test on another `0081` sensor once we have a procedure that accounts for the existing Windows fingerprints. We will seek the owner's consent before any test that changes the device or system.