uunicorn/python-validity #42: RPM package

mavahedinia, 2020-09-21:
Hey there and thanks for your works.

can you provide rpm packages as well as deb package provided? or help us install this library on fedora/opensuse?

RaghavRao, 2020-10-16:
Hi, I am using this on Fedora 32 with no issues.
I git cloned this repo and https://github.com/uunicorn/open-fprintd repo, did `sudo pip install .` then copied over the files from the debian folder into the correct locations (system services into /etc/systemd/system etc)

Another option is to download the .deb package from the ppa and use the alien program to convert deb to rpm

If you run into an error that the package conflicts with file system  when using alien read [this](https://www.electricmonk.nl/log/2017/02/23/how-to-solve-rpms-created-by-alien-having-file-conflicts/)

Again, I can give more step by step instructions if you get stuck.

An official RPM would be nice however

ArthurNagy, 2020-10-24:
Hey, @RaghavRao I'd really appreciate a more step by step instruction, I'm being a complete noob and can't figure out how to properly install it on Fedora.

RaghavRao, 2020-10-28:
On Fedora:

$ cd ~
$ git clone https://github.com/uunicorn/python-validity/
$ cd python-validity
$ sudo cp debian/python3-validity.service /etc/systemd/system/
$ sudo cp debian/python3-validity.udev /etc/udev/rules.d/10-validity.rules
$ sudo cp dbus_service/io.github.uunicorn.Fprint.conf
/usr/share/dbus-1/system.d/io.github.uunicorn.Fprint.conf
$ sudo cp dbus-service/dbus-service
/usr/local/lib/python-validity/dbus-service
$ sudo pip install .
$ sudo systemctl enable python3-validity.service
$ sudo systemctl start python3-validity.service
$ sudo systemctl status python3-validity.service
$ fprintd-enroll

alexjfinch, 2020-11-02:
> On Fedora: $ cd ~ $ git clone https://github.com/uunicorn/python-validity/ $ cd python-validity $ sudo cp debian/python3-validity.service /etc/systemd/system/ $ sudo cp debian/python3-validity.udev [...]

Hi - I've followed this, however I'm getting the following error when trying to run the service...

● python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/etc/systemd/system/python3-validity.service; static)
     Active: failed (Result: exit-code) since Mon 2020-11-02 21:04:35 GMT; 16s ago
    Process: 159302 ExecStart=/usr/lib/python-validity/dbus-service --debug (code=exited, status=1/FAILURE)
   Main PID: 159302 (code=exited, status=1/FAILURE)
        CPU: 152ms

Nov 02 21:04:35 galahad dbus-service[159302]: Traceback (most recent call last):
Nov 02 21:04:35 galahad dbus-service[159302]:   File "/usr/lib/python-validity/dbus-service", line 307, in <module>
Nov 02 21:04:35 galahad dbus-service[159302]:     main()
Nov 02 21:04:35 galahad dbus-service[159302]:   File "/usr/lib/python-validity/dbus-service", line 260, in main
Nov 02 21:04:35 galahad dbus-service[159302]:     backoff()
Nov 02 21:04:35 galahad dbus-service[159302]:   File "/usr/lib/python-validity/dbus-service", line 214, in backoff
Nov 02 21:04:35 galahad dbus-service[159302]:     with open(backoff_file, 'w') as f:
Nov 02 21:04:35 galahad dbus-service[159302]: FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/python-validity/backoff'
Nov 02 21:04:35 galahad systemd[1]: python3-validity.service: Main process exited, code=exited, status=1/FAILURE
Nov 02 21:04:35 galahad systemd[1]: python3-validity.service: Failed with result 'exit-code'.

I couldn't find anything for "backoff" in the download to put in that location.

Apologies, I'm a complete noob!

veitw, 2020-11-14:
You might want to have a look at the RPM packages provided in issue #54.