uunicorn/python-validity #45: python-validity used in sudo allow root escalation without passwd or fingerprint

ptonelli, 2020-10-05:
if an unauthorized person can get accessed to an unlocked laptop, the person can gain sudo by replacing the existing fingreprints by its own without requiring authentication. 

Steps to reproduce:
fprintd-delete
fprintd-enroll [put attacker's fingerprints]
sudo su - [put attacker's fingerprints]
root \o/

This looks similar to [this bug](https://bugs.launchpad.net/ubuntu/+source/fprintd/+bug/1532264) in the original fprintd.

RaghavRao, 2020-10-16:
Would the best option be requiring password for fingerprint enroll?

ptonelli, 2020-10-16:
Not sure my understanding of the thread in the link is perfect, but I believe three things are necessary:
- requiring sudo or the current password to enroll fingerprints
- blocking the user from writing to the fingerprint files. (chown root ?). 
- checking the file permissions and ownership before each fingerprint check (else you might be able to move the folder containing the file and replace it by a user owned file)

haselwarter, 2020-11-07:
The `python3-validity` script is usually run as a systemd service, under the root user. `fprintd-enroll` connects to `open-fprintd` via D-Bus which in turn connects to `python3-validity`. This allows any user to send commands to a process run as root. The `open-fprintd` project page does indeed contain a warning:
>  At this point this project does not enforce any auth checks against requests made on DBus.
Such checks would help avoiding this exploit.