uunicorn/python-validity #46: Fingerprint recognition unreliable on 009a

mlalkaka, 2020-10-06:
First off, thank you so much for working on this!

I have a ThinkPad T480s with the Validity 009a fingerprint reader. I enrolled and registered my fingerprint fine, but I've noticed that recognizing the fingerprint doesn't seem to be working very well. It only recognizes the fingerprint very rarely. I tried re-registering the fingerprint and tried multiple different finger positions, but I got the same result.

Is this a known issue? Any tips on how I can improve this? Let me know if I can provide more info to diagnose the issue.

himekifee, 2020-10-06:
You can try do a factory reset to achieve a calibration, not sure if that works for you..

mlalkaka, 2020-10-06:
How do I do a factory reset?

mlalkaka, 2020-10-06:
Oh, found what you were referring to: https://github.com/uunicorn/python-validity#error-situations. I'll give that a try

uunicorn, 2020-10-17:
> Is this a known issue? Any tips on how I can improve this?

I had similar issues when I was accidentally holding my palm against the sensor while it was calibrating.
I was able to recover from this state by deleting the `/usr/share/python-validity/calib-data.bin` file and performing a factory reset.

elmir-becirovic, 2020-10-20:
I am facing the same issue as mentioned here. I have noticed that if I tap my finger really fast on the scanner it manages to match it, but if I am a bit slower it will fail to match. Any ideas?

mlalkaka, 2020-10-28:
Thanks @himekifee and @uunicorn . Factory resetting as described in [Error Situations](https://github.com/uunicorn/python-validity#error-situations) worked! The fingerprint sensor has been reliable and pretty fast now. Closing this issue.

closed 2020-10-28 by mlalkaka