uunicorn/python-validity #47: Thinkpad T480: list_devices failed:

kendoori, 2020-10-14:
Followed install procedure on Pop!_OS 20.04 and received:

```
$ fprintd-enroll
list_devices failed: 
```
I then tried to reset the firmware per below.

```
sudo python3 /usr/share/python-validity/playground/factory-reset.py
[sudo] password for kenny: 
Traceback (most recent call last):
  File "/usr/share/python-validity/playground/factory-reset.py", line 5, in <module>
    usb.open()
  File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 49, in open
    self.open_dev(dev)
  File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 61, in open_dev
    raise Exception('No matching devices found')
Exception: No matching devices found

```

oshank, 2020-10-14:
What is the output of `lsusb`?

kendoori, 2020-10-14:
```
lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0bda:0316 Realtek Semiconductor Corp. USB3.0-CRW
Bus 002 Device 005: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 004: ID 05e3:0620 Genesys Logic, Inc. 
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 5986:2113 Acer, Inc Integrated Camera
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 011: ID 04f9:60e0 Brother Industries, Ltd 
Bus 001 Device 008: ID 047d:2048 Kensington Orbit Trackball with Scroll Ring
Bus 001 Device 015: ID 04d9:a0cd Holtek Semiconductor, Inc. 
Bus 001 Device 014: ID 0b0e:0349 GN Netcom 
Bus 001 Device 012: ID 046d:0892 Logitech, Inc. OrbiCam
Bus 001 Device 009: ID 08bb:29b6 Texas Instruments PCM2906B Audio CODEC
Bus 001 Device 006: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 013: ID 04f2:0976 Chicony Electronics Co., Ltd 
Bus 001 Device 010: ID 0c76:161e JMTek, LLC. 
Bus 001 Device 016: ID 0b0e:2467 GN Netcom USB2.0 Hub
Bus 001 Device 007: ID 0922:0022 Dymo-CoStar Corp. 
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

kendoori, 2020-10-14:
My fingerprint reader was not enabled in BIOS. I reenabled and it is now visible :-)

closed 2020-10-14 by kendoori