uunicorn/python-validity #49: Failed to show user's enrolled fingerprints on ThinkPad L480/Ubuntu 20.04.1 LTS

stz184, 2020-10-20:
I am using ThinkPad L480 which has the same fingerprint reader as T480 and T480s (06cb:009a)
I was able to add a finerprint using `fprintd-enroll` but failed to do so using the Gnome interface. 
The recognition works as expected, but I have problem listing the enrolled fingerprints both using `fprintd-list` and Gnome interface (Settings -> Users -> Fingerpint login).

I am getting the following output when I try to list my fingerpints:

```
vladimir@ThinkPad-L480:~$ fprintd-list stz184
found 1 devices
Device at /net/reactivated/Fprint/Device/1
Using device /net/reactivated/Fprint/Device/1
ListEnrolledFingers failed: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 711, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python-validity/dbus-service", line 90, in ListEnrolledFingers
    raise e
  File "/usr/lib/python-validity/dbus-service", line 81, in ListEnrolledFingers
    usr = self.user2record(user)
  File "/usr/lib/python-validity/dbus-service", line 64, in user2record
    return db.lookup_user(self.user2identity(user))
  File "/usr/lib/python-validity/dbus-service", line 54, in user2identity
    pw = pwd.getpwnam(user)
KeyError: "getpwnam(): name not found: dbus.String('stz184')"

```

I am using the latest available version via the PPA.

stz184, 2020-10-23:
P.S. The same version works flawlessly on Linux Mint 20 Ulyana (based on Ubuntu 20.04 LTS)

JMendyk, 2020-11-04:
`fprintd-list` expects username as the second argument. The error message you provided, based on my attempt to reproduce it, may happen if user with provided username does not exist.

Based on your listing I guess you should invoke `fprintd-list` as `fprintd-list vladimir`.

stz184, 2020-11-11:
@JMendyk thanks for pointing me out the mistake, but this does not resolve the issue completely:
- If fprintd-list command is issued without username as a second parameter, the above-mentoned error is thrown (the same error as when a wrong username is supplied). The same case on Linux Mint leads to a nice help message "Usage: fprintd-list <username> [usernames...]"
- The gnome settings panel also fails to display the enrolled fingers - I can successfully list my enrolled fingers using `fprintd-list vladimir` command but I can't view them using Settings -> Users -> Fingerpint login

JMendyk, 2020-11-11:
@stz184 Since that's the case, unfortunately I don't think I can help you in any way.

As a help for someone more familiar with the code base, on my machine -- also ThinkPad L480, running Ubuntu 18.04.5, `fprintd-list` properly prints usage information and gnome settings panel recognises that I have enrolled fingers.

stz184, 2020-12-05:
I just upgraded to Ubuntu 20.10 and reinstalled  open-fprintd fprintd-clients python3-validity. The result is pretty much the same - `frpintd-list` shows proper list of enrolled fingerprints but Gnome GUI fails. 
See the screenshot: https://imgur.com/a/9oLICGC