uunicorn/python-validity #51: After enrolling finger-print, how to modify the settings in /etc/pam.d/ folder?

ipstone, 2020-10-31:
Hello,

Thanks for the nice work on python-validity. I have it installed successfully on manjaro/arch - as I couldn't find pam-auth-update command in arch, I am trying to modify the content in the /etc/pam.d/ folder manually:

Here are the files I have in the `/etc/pam.d/` folder:
```
chage
chfn
chgpasswd
chpasswd
chsh
cinnamon-screensaver
crond
cups
groupadd
groupdel
groupmems
groupmod
i3lock
lightdm
lightdm-autologin
lightdm-greeter
login
newusers
other
passwd
polkit-1
rlogin
rsh
runuser
runuser-l
samba
shadow
sshd
su
sudo
su-l
system-auth
systemd-user
system-local-login
system-login
system-remote-login
system-services
useradd
userdel
usermod
vlock
```

As I am using lightdm as my gui login, I first tried to add the 'changed lines' in the README file to the end of `lighdm` file, the result is the following: 
```
#%PAM-1.0
auth        include     system-login
-auth       optional    pam_gnome_keyring.so
account     include     system-login
password    include     system-login
session     include     system-login
-session    optional    pam_gnome_keyring.so auto_start

# In /etc/pam.d/common-auth, the following line is added, and the next line changed.
# The end result (apart from other things that may be in the file) is this:
auth  [success=2 default=ignore]  pam_fprintd.so max_tries=1 timeout=10 # debug
auth  [success=1 default=ignore]  pam_unix.so nullok_secure try_first_pass
```

Probably due to my lack of understanding how pam works - this setup caused my login in a loop / not working: after trying to enter the password in the text area of lightdm, then the fingerprint indicator lights up - but using my finger print, it would not login (even with the right password in the text box). It's kind of a loop - for which I suspect is the sequences I have in the `lightdm` file are wrong.

What I hope to setup is: 1. use finger-print to login, 2. use finger-print to unlock from screen lock.
I would really appreciate if you have suggestion/answers on how to archieve this in arch/manjaro linux.

thanks!

jjgalvez, 2020-12-16:
After lots of googling and lots of trial and error I got mine working, You need to edit system-auth. I added 
`auth       [success=3 default=ignore]  pam_fprintd.so                                  # added to use the fingerprint reader`
above 
`auth       [success=2 default=ignore]  pam_unix.so          try_first_pass nullok`
this way it checks the FP reader first.  You can switch the lines (if you do you need to also switch the success=x so that the first one is 3 and the second one is 2) so that it asks for a password first but I like it the other way around

rmeissn, 2021-06-16:
Alternatively see https://github.com/uunicorn/python-validity/pull/82#issuecomment-862411183