uunicorn/python-validity #55: Exception: Unexpected TLS version 4 0

MGodric, 2020-11-16:
I'm trying to apply your work on a T470p with 138a:0097 sensor.

And this exception jumped out when I tried to initialize the device, as below:

```
Found device <DEVICE ID 138a:0097 on Bus 001 Address 003>
Downloading https://download.lenovo.com/pccbbs/mobiles/n1mgf03w.exe to extract 6_07f_lenovo_mis.xpfwext
Extracting "Synaptics Metallica MOC Touch Fingerprint Reader Driver for Windows 8.1 64-bit" - setup data version 5.5.0
 - "app/WBF_Drivers/6_07f_lenovo_mis.xpfwext"
Done.
Found firmware at /tmp/tmp5tk0_boc/app/WBF_Drivers/6_07f_lenovo_mis.xpfwext
The device will be now reset to factory and associated to the current laptop.
Press Enter to continue (or Ctrl+C to cancel)...
Opening device 0x97
Factory reset...
Sleeping...
Pairing the sensor with device 20J6A019CD
Opening device 0x97
Initializing flash...
Detected Flash IC: W25Q80B, 1048576 bytes
Sleeping...
Try 1 failed with error: Failed: 04af
Opening device 0x97
Initializing flash...
Sleeping...
Try 2 failed with error: Unexpected TLS version 4 0
Opening device 0x97
Initializing flash...
Sleeping...
Try 3 failed with error: Unexpected TLS version 4 0
Opening device 0x97
Initializing flash...
Sleeping...
Try 4 failed with error: Unexpected TLS version 4 0
Opening device 0x97
Initializing flash...
Sleeping...
Try 5 failed with error: Unexpected TLS version 4 0
Device didn't reply in time...
Traceback (most recent call last):
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 339, in <module>
    vfs_tools.initialize(fwpath, args.calibration_data)
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 249, in initialize
    self.pair(fwpath, calib_data)
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 220, in pair
    self.retry_command(init_flash_command, max_retries=5)
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 112, in retry_command
    raise(err)
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 103, in retry_command
    command()
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 219, in init_flash_command
    init_flash()
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/init_flash.py", line 97, in init_flash
    partition_flash(flash_layout_hardcoded, client_public)
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/init_flash.py", line 70, in partition_flash
    info = get_flash_info()
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/flash.py", line 28, in get_flash_info
    rsp=tls.cmd(unhex('3e'))
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/tls.py", line 104, in cmd
    rsp=self.app(cmd)
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/tls.py", line 138, in app
    return self.parse_tls_response(self.usb.cmd(self.make_app_data(b)))
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/tls.py", line 336, in parse_tls_response
    raise Exception('Unexpected TLS version %d %d' % (mj, mn))
Exception: Unexpected TLS version 4 0
```
Thank you for your work!

uunicorn, 2020-11-16:
It looks like you are using @3v1n0 's pairing tool which is based on outdated version of `python-validity`. You don't need it for `138a:0097`.

MGodric, 2020-11-17:
Oh, I tried updating the python-validity and reset the sensor. It works!

Thank you! Nice work!

closed 2020-11-17 by MGodric

spandansinha68, 2021-06-04:
im having the exactly same problem on the 0090 sensor on t460s when i tried to initialize the device

Luism2600, 2022-04-06:
> Oh, intenté actualizar la validez de Python y restablecer el sensor. ¡Funcio

> Oh, I tried updating the python-validity and reset the sensor. It works!
> 
> Thank you! Nice work!

podrías comentarnos como lo hiciste?