uunicorn/python-validity #59: Synaptics 06cb:009a on ThinkPad T480 Not Functioning After Waking From Sleep

ryangwsimmons, 2020-12-03:
Hi,

The title pretty much sums up the issue. I'm running Manjaro, using kernel 5.4.80, and KDE Plasma 5.20.4. I installed python-validity 0.12-2 yesterday, and am very happy to confirm that under most circumstances, it's working great for me. I can log out or lock my laptop, and the fingerprint reader works as expected. Thank you very much for all of your work on this. 

However, there's one issue I'm having. After I put my laptop into sleep, and then wake it, the fingerprint reader no longer works. Typically, to use the fingerprint reader, I would hit enter on the empty password field in SDDM or the KDE lock screen, which would cause the light above the fingerprint reader to turn on. After this, I could put my finger on the reader and it would work. However, in this scenario, the light above the fingerprint reader doesn't even turn on. SDDM/the lock screen just immediately says "invalid password". After this, it no longer works until I reboot my laptop.

Running `sudo systemctl status python3-validity.service` after waking from sleep produces the following output:

```
● python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/usr/lib/systemd/system/python3-validity.service; static)
     Active: active (running) since Thu 2020-12-03 18:18:33 EST; 26min ago
   Main PID: 714 (python3)
      Tasks: 3 (limit: 18891)
     Memory: 28.2M
     CGroup: /system.slice/python3-validity.service
             └─714 python3 /usr/lib/python-validity/dbus-service --debug

Dec 03 18:18:53 Ryan-T480 python3[714]: DEBUG:root:<tls< 17: 00008a0001000400040000000300020002000400200053bc96ce1a6dccbee9765cef44a71bf1648f5e3f38b835c9863f81f466a9f53e05002000000000000000000000000000000000000000000000000000000000000000000002000400c10300000600280000000000000000000000000000000000000000000000000000000000000000000000000000000000
Dec 03 18:18:53 Ryan-T480 python3[714]: DEBUG:root:>tls> 17: 6200000000
Dec 03 18:18:53 Ryan-T480 python3[714]: DEBUG:root:>cmd> 1703030040b2b2e5f93cd236452803ff3cbccadc6f7bb0cf8bc92e51151db97829096c3eca357bf6c969053e65de42c4d21e37f7fe0d7298b9b08f54e96e0104ba17d7211a
Dec 03 18:18:53 Ryan-T480 python3[714]: DEBUG:root:<cmd< 170303004098ee187d5eb5d26a126d2357308fbdcc77c3eaf7442e0c768eee1678239b793cc51d746c665821ff397749ce23abb2f4f6f25c62e6355465ae17bf2fdbeb6e9c
Dec 03 18:18:53 Ryan-T480 python3[714]: DEBUG:root:<tls< 17: 0000
Dec 03 18:18:53 Ryan-T480 python3[714]: DEBUG:root:VerifyStatus
Dec 03 18:42:56 Ryan-T480 python3[714]: DEBUG:root:In ListEnrolledFingers ryan
Dec 03 18:42:56 Ryan-T480 python3[714]: DEBUG:root:>tls> 17: 4b00000b0053746757696e64736f7200
Dec 03 18:42:56 Ryan-T480 python3[714]: DEBUG:root:>cmd> 17030300505d2db5521a6ef042688919fe9339f2e8aa3cfb047b6a29f9bd7f7c3adddb65a2ac227b5c4b807fe7fa7c2d7d3af0fa24d5aeb659c77405c5d13866eaf0799ab7119ee2f2774e6082a4b3c190fba74179
Dec 03 18:42:56 Ryan-T480 python3[714]: DEBUG:root:<cmd< 0404
```

Please let me know if there's any other information you need. Thanks.

uunicorn, 2020-12-07:
Hi,

It looks like the the `python-validity` driver has missed the system suspend notification.
Can you also check the statuses of  `open-fprintd.service`, `open-fprintd-resume.service` and `open-fprintd-suspend.service`?

ryangwsimmons, 2020-12-08:
Ah ha! Looks like `open-fprintd-resume.service` and `open-fprintd-suspend.service` were disabled. After enabling them, rebooting, and trying to put the computer into sleep again, the fingerprint reader now works after waking. Thanks.