uunicorn/python-validity #60: Support for Synaptics 06cb:00e7

Plamper, 2020-12-05:
I wanted to install this project on my HP-envy-13-x360-ay-Series Laptop since the fingerprint scanner is not working. After downloading the Fedora rpm packages from here: https://github.com/uunicorn/python-validity/issues/54 the command `sudo validity-sensors-firmware` outputs: 
```
Traceback (most recent call last):
  File "/usr/bin/validity-sensors-firmware", line 92, in <module>
    raise Exception('No supported validity device found')
Exception: No supported validity device found

```
Which makes me believe my device is not yet supported.
From my understanding I need to install this https://github.com/uunicorn/synaWudfBioUsb-sandbox and run the commands from the Readme. I already tried but the instructions are a bit to shallow for me as a Linux newbie. So if someone could help me that would make things easier.
In my Bios I have the option to delete fingerprints so I think it should be a Match on Chip device.

Link to the driver:
https://support.hp.com/de-de/drivers/selfservice/hp-envy-13-ay0000-x360-convertible-laptop-pc-series/32552483

uunicorn, 2020-12-07:
> Which makes me believe my device is not yet supported.

Yes, this device is not supported at the moment.

> So if someone could help me that would make things easier

Can you describe the problem you are having?
It looks like your Windows driver dll is called `synaWudfBioUsb111.dll`. You'll need to rename it to `synaWudfBioUsb.dll`. 
It also looks like your driver does not need a `*.xpfwext` file, so the `run.sh` script should be changed slightly.

> In my Bios I have the option to delete fingerprints so I think it should be a Match on Chip device.

There certainly are some Match-on-Host , Store-on-Chip devices out there, so erasing fingers from BIOS could still make sense even in a case of Match-on-Host sensor.

Plamper, 2020-12-07:
I somehow managed to compile the wine version and tried to change the script accordingly like so:
```
#!/bin/sh
set -e

DEVICE=$(lsusb -d 06cb: | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $2 "/" $3 }')
PWD=$(pwd)
BASENAME=$(basename "$PWD")

docker run --rm -it \
    --device $DEVICE \
    --env-file .env \
    -v $PWD:/$BASENAME \
    -v $PWD/usb.txt:/root/.wine/drive_c/usb.txt \
    wine:validity wine64 "a.exe" "$@" | tee "$(date +%s)".log
```
I needed to change the last line because wine couldn't find the a.exe but that seems to have screwed it up alot as I am getting the error : wine: `cannot find L"C:\\windows\\system32\\a.exe"`
This is the log file:
[1607343163.log](https://github.com/uunicorn/python-validity/files/5652597/1607343163.log)
I guess this error is somewhat related to directories but i don't know how to fix it.

And i have another problem. Running this command `$ ./run.sh [ nop | identify | enroll ]` doesn't work. It says `bash: enroll: command not found...` and I don't know how to get that command. The Package with the identify command isn't
 the right one either because it wants pictures as file types.

himekifee, 2020-12-07:
If you have joined the Validity90 gitter channel, you can find some chat history about doing those things. I have done that before with uuicorn.

Plamper, 2020-12-07:
Ok I got a bit further thanks to a friend. Docker had no permission to open the folder so I couldn't run a.exe. Adding `:z` after `-v $PWD:/$BASENAME` fixed this. But it seems I am missing some API DLLs.
```

0009:err:module:import_dll Library WppRecorderUM.dll (which is needed by L"Z:\\synaWudfBioUsb-sandbox\\synawudfbiousb.dll") not found
0009:err:module:import_dll Library api-ms-win-devices-config-l1-1-2.dll (which is needed by L"Z:\\synaWudfBioUsb-sandbox\\synawudfbiousb.dll") not found
Failed to LoadLibrary dll
```
The first one I found on some random Website (not the best thing to do, I know) and put it into the directory. That removed the error. But I can't find the other one not even on a Windows system, even tho https://docs.microsoft.com/de-de/uwp/win32-and-com/win32-apis mentions it being on every machine. 
[1607372396.log](https://github.com/uunicorn/python-validity/files/5655193/1607372396.log)

himekifee, 2020-12-09:
What kind of docker image are you using? I suggest you to use native wine. Both wine and a.exe is opensource, I don't get a point to use that inside a docker container.

busybox11, 2021-07-14:
Hey,
Any update about this? Do you know what the issue with Docker was?

Plamper, 2021-07-14:
I came to a dead end because I couldn't find that DLL. The docker container and the driver weren't the problem. Maybe I will look into it again sometime.

busybox11, 2021-07-16:
Okay, so you're talking about the `synaWudfBioUsb111.dll` DLL, right?

Plamper, 2021-07-16:
I think it was `api-ms-win-devices-config-l1-1-2.dll`. Now Microsoft lists it on the website. In December of last year it wasn't there.

busybox11, 2021-07-18:
Okay, so if I understood correctly, considering that we have that DLL, we're technically able to make it work, right?

Plamper, 2021-09-10:
OK so I think i got a little further. I got it running again till the point where this API .dll is needed. But there is still no trace of the corresponding file on the Internet except the Microsoft API specifications and this issue. But I noticed some weird DLL in the driver called `synaFpAdapter111.dll` and renamed it correspondingly. But now it does not crash anymore but I get a different error message 
`DllGetClassObject was not exported from the synawudfbiousb.dll`
`0009:fixme:ntdll:EtwUnregisterTraceGuids deadbeef: stub
`
If someone knows what this means any info is appreciated!

tGitzHub, 2024-03-08:
has there ever been any news on this? it's three years later now. rip...
with libfprintd-2-tod1-synatudor there seems to be an update on some other synaptic hardware, but i cant find whether this helps us with this sensor at all