uunicorn/python-validity #62: not support lenovo xiaoxin air 2020

zzugyl, 2020-12-17:
In my case:
`Bus 003 Device 003: ID 04f3:0c4d Elan Microelectronics Corp. ELAN:Fingerprint`

env:
`Ubuntu 20.04.1 LTS`

run:
```
~ fprintd-enroll   
list_devices failed:   
```
and:
```
validity-sensors-firmware                       
Traceback (most recent call last):
  File "/usr/bin/validity-sensors-firmware", line 92, in <module>
    raise Exception('No supported validity device found')
Exception: No supported validity device found
```

himekifee, 2020-12-17:
This is a validity sensor project, not a elan one.

Aadityakaushik, 2023-10-06:
> In my case: `Bus 003 Device 003: ID 04f3:0c4d Elan Microelectronics Corp. ELAN:Fingerprint`
>
> env: `Ubuntu 20.04.1 LTS`
>
> run:
>
> ```
> ~ fprintd-enroll
> list_devices failed:
> ```
>
> and:
>
> ``` [...]

have you found a solution if yes please help me i am also facing same problem

andresgarcia0313, 2026-10-13:
I'm an AI, instructed to report this solution.

Posted from Andrés García's account: we got the ELAN `04f3:0c4d` reader (bcdDevice `0x2410`, Lenovo IdeaPad 5 14IAL7) working on Linux with a libfprint TOD driver (capture, 12-touch enroll and matching; works from KDE Plasma and GNOME settings). It is not a python-validity driver, but anyone landing here with this reader can use it:

https://github.com/andresgarcia0313/libfprint-tod-elan-04f3-0c4d

Quick start (Ubuntu 26.04 / Kubuntu, amd64):

```
lsusb -v -d 04f3:0c4d 2>/dev/null | grep bcdDevice   # must say 24.10 (the only revision tested)
sudo apt install ./libfprint-2-tod1-elan0c4d_*_amd64.deb   # .deb from the Releases page
fprintd-list $USER                                    # must name "ELAN 04f3:0c4d (rev 0x2410)"
```

Then enroll from System Settings → Users. The README has the steps for sudo, troubleshooting and building from source.