uunicorn/python-validity #63: Debian Buster

cgarlati, 2020-12-19:
Is there an installation package for Debian Buster 10.6 ?

niklasarnitz, 2021-11-01:
Have a look at the Debian repository provided by @uunicorn.
You can add it to Debian 11 by using the following APT-sources line:
```
deb http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu impish main
```
But please keep in mind that you should only set the ubuntu release of the repository to the "corresponding" version of your Debian installation.

CornelisDenhart, 2022-10-30:
For me to get it working on Debian 11 Bullseye on a Lenovo T480, I had to add this to /etc/apt/sources.list
`deb http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu jammy main` 
Then do
`apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 8D4C774BA6D18F90
apt-get update
apt-get install open-fprintd fprintd-clients python3-validity
fprintd-enroll ...`

bradmcgo, 2023-08-26:
> For me to get it working on Debian 11 Bullseye on a Lenovo T480, I had to add this to /etc/apt/sources.list `deb http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu jammy main` Then do `apt-key [...]

This works for me on Debian 12 with device ID 06cb:009a

fourthperson, 2023-09-20:
> This works for me on Debian 12 with device ID 06cb:009a

Works for me too on Debian 12 and device 06cb:009a

charlesvill, 2023-10-04:
> Works for me too on Debian 12 and device 06cb:009a

How do we add the  `deb http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu jammy main` line to sources.list? its read-only for me?

charlesvill, 2023-10-04:
never mind, fixed it. Nothing a little sudo vim can't fix ;)

ddr5ecc, 2024-01-08:
Thank you so much. 
I did it on 138a:0097, Debian 11, Thinkpad X270

JamesNathan1, 2024-06-18:
I'm using debian 12, i got errors  fprintd-clients : Depends: libc6 (>= 2.38) but 2.36-9+deb12u7 is to be installed.  "sudo apt install libc6:>=2.38" seems to fix it. i'm using the jammy repo.