uunicorn/python-validity #64: TLS connection fails after suspend (lid closed)

xenomorph1096, 2020-12-20:
Fingerprint sensor works fine before suspend (both `systemctl suspend` and closing the lid).
After waking up from suspend, the validity service stops accepting TCP connections and I am unable to use the sensor for any pam authentication. After suspend, `fprintd-list <user>` returns the following error:

```
❯ fprintd-list user
found 1 devices
Device at /net/reactivated/Fprint/Device/1
Using device /net/reactivated/Fprint/Device/1
ListEnrolledFingers failed: Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/dbus/service.py", line 711, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python-validity/dbus-service", line 90, in ListEnrolledFingers
    raise e
  File "/usr/lib/python-validity/dbus-service", line 81, in ListEnrolledFingers
    usr = self.user2record(user)
  File "/usr/lib/python-validity/dbus-service", line 64, in user2record
    return db.lookup_user(self.user2identity(user))
  File "/usr/lib/python3.9/site-packages/validitysensor/db.py", line 163, in lookup_user
    stg = self.get_user_storage(name='StgWindsor')
  File "/usr/lib/python3.9/site-packages/validitysensor/db.py", line 149, in get_user_storage
    return parse_user_storage(tls.cmd(pack('<BHH', 0x4b, dbid, len(name)) + name))
  File "/usr/lib/python3.9/site-packages/validitysensor/tls.py", line 122, in cmd
    rsp = self.app(cmd)
  File "/usr/lib/python3.9/site-packages/validitysensor/tls.py", line 152, in app
    return self.parse_tls_response(self.usb.cmd(self.make_app_data(b)))
  File "/usr/lib/python3.9/site-packages/validitysensor/tls.py", line 351, in parse_tls_response
    raise Exception('Unexpected TLS version %d %d' % (mj, mn))
Exception: Unexpected TLS version 4 0

```

Issue no longer persists after rebooting or restarting the python3-validity systemd service.

Distro: Archlinux 64-Bit
Laptop: Lenovo X1 Carbon (Gen6)

Other than this, kudos. Works very well indeed.

daanjderuiter, 2020-12-22:
I thought I was experiencing this is well, but then came across #59, maybe that's affecting you as well? Starting and enabling `open-fprintd-resume.service` and `open-fprintd-suspend.service` did the trick for me.

xenomorph1096, 2020-12-23:
Yes! that did it. I didn't see that other issue~ Thanks

closed 2020-12-23 by xenomorph1096