uunicorn/python-validity #67: Void linux / non systemd support

KoalaV2, 2020-12-26:
Please add support for non systemd linux distros!

VPetukhov, 2021-01-25:
Yes, please!

SimPilotAdamT, 2021-12-05:
To add something a bit more constructive, it seems like the `open-fprintd`'s dependency of `systemd-libs` is the reason why this doesn't support distro's like Artix/Void Linux.

Midou36O, 2023-08-17:
I got it to work 
[image]

The steps that worked for me was:
1. Manually clone the repo to `/opt/python-validity`
    * Go to the repo and run `python setup.py install --prefix=/usr`
2. Get the [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=open-fprintd) of open-fprintd
3. Remove [`fprintd-clients`](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=open-fprintd#n9) (line 9) from the dependencies.
4. Run `makepkg -si` and let it install the "open-fprintd" package.
5. Write an init script for your init of your choice, in my case it was OpenRC:
     * Filename: open-fprintd
```bash
#!/usr/bin/openrc-run                                                                        
                                                                                             
supervisor=supervise-daemon                                                                  
command="env PYTHONPATH=/opt/python-validity /opt/python-validity/dbus_service/dbus-service" 
command_args=""                                                                              
pidfile=                                                                                     
                                                                                             
name="open-fprintd"                                                                          
description="Open FPrint Daemon"                                                             
```
   - (This is very rough so please tell me the proper stuff to fix my init script file.)
6. Place the init file in `/etc/init.d` and `chmod +x` it
7. Install fprintd (`pacman -S fprintd`)
8. Start the init file you just made using `rc-service open-fprintd start`
9. If it works, add it to your default init runlevel `rc-update add open-fprintd`
10. Run `fprintd-enroll` and hope it works :)

Everything should be done as root with some obvious exceptions.

Kartonowy, 2024-12-11:
Any update on any of this? Somebody figured it out on Void?