uunicorn/python-validity #69: Can't install with yay from Arch AUR

apl1341, 2020-12-30:
I've been trying to install this driver on my ThinkPad T480s but I keep running into the same error where yay cant find the dependencies for it. I also tried installing the deps from git directly with no luck.

Going to the AUR webpage and looking for fprintd-clients and open-fprintd just gives a 404.

See below:

`~ ❯ yay -S python-validity-git                                                                                                                              3s   09:18:09 PM
 -> Could not find all required packages:
        fprintd-clients (Wanted by: python-validity-git)
        open-fprintd (Wanted by: python-validity-git)
  ~ ❯                                                                                                                                                         4s   09:20:30 PM`

KindlyPizza, 2020-12-30:
Same, just reloaded Arch on my machine and suddenly found that dependencies were missing.

jugendhacker, 2020-12-30:
Yeah I also searched aur-request mailing list for anything about what's going on but the package just disappeared.... I'll maybe ask on aur-general

jugendhacker, 2020-12-30:
Okay so I asked on aur-general and they told me the packages where removed because the maintainer deleted their account. But luckily the git for the package is still there and so I republished them.

KindlyPizza, 2020-12-30:
many thanks upon you.

apl1341, 2020-12-30:
Thanks so much! Just got it working on my machine.

closed 2020-12-31 by apl1341