uunicorn/python-validity #71: fprintd-enroll fails with 138a:0090 ?

dlarrick, 2021-01-09:
Greetings,

I've been successfully using Marco Trevisan's [PPA](https://launchpad.net/~3v1n0/+archive/ubuntu/libfprint-vfs0090) for my Thinkpad on Debian (latest testing/bullseye), initialized & flashed last spring by his fork of this project. But that PPA seems to have gone unmaintained and is no longer installable after a Debian fprintd migration, so I figured I would give this project a try.

Install went fine, either from the PPA or from a self-built python3-validity package. 

It warned that the sensor was already paired to another OS so I did the validity-sensors-firmware / factory-reset.py thing. But when I try to enroll a finger it immediately fails:
```
$ fprintd-enroll 
Using device /net/reactivated/Fprint/Device/1
Enrolling right-index-finger finger.
Enroll result: enroll-failed
```
Logs show:
```
Jan 09 08:39:24 rocket python3[46186]: DEBUG:root:EnrollStatus
Jan 09 08:43:03 rocket python3[46186]: DEBUG:root:In EnrollStart right-index-finger for doug
Jan 09 08:43:03 rocket python3[46186]: DEBUG:root:>tls> 17: 6901000000
Jan 09 08:43:03 rocket python3[46186]: DEBUG:root:>cmd> 17030300409eefbc642369b985f8134533b7a25dc89342c0a84312fa31210735931a022c5e7d34928b89cb8e9bd37>
Jan 09 08:43:03 rocket python3[46186]: DEBUG:root:<cmd< 1703030040cc2d615ce8d700682dbbf9992cc03be04e967a8fe356a96d4b523a14f4f7602e500f45602c3f78f96e5>
Jan 09 08:43:03 rocket python3[46186]: DEBUG:root:<tls< 17: 0104
Jan 09 08:43:03 rocket ERROR:root:Failed[46186]: 0401
                                                 Traceback (most recent call last):
                                                   File "/usr/lib/python-validity/dbus-service", line 161, in run
                                                     sensor.enroll(usr, index, update_cb)
                                                   File "/usr/lib/python3/dist-packages/validitysensor/sensor.py", line 828, in enroll
                                                     self.create_enrollment()
                                                   File "/usr/lib/python3/dist-packages/validitysensor/sensor.py", line 749, in create_enrollment
                                                     assert_status(tls.app(pack('<BL', 0x69, 1)))
                                                   File "/usr/lib/python3/dist-packages/validitysensor/util.py", line 12, in assert_status
                                                     raise Exception('Failed: %04x' % s)
                                                 Exception: Failed: 0401
Jan 09 08:43:03 rocket python3[46186]: DEBUG:root:EnrollStatus
```
I have a hard time believing support for this first Validity sensor is broken so I must have missed a step or be doing something else wrong. But I'm at a loss. Any guidance or suggestions would be appreciated

himekifee, 2021-01-09:
Did you pair the sensor with pc use validity tools?

dlarrick, 2021-01-09:
Thanks for the suggestion. My understanding is the dbus-service does pairing automatically at startup if necessary.

If I manually run validitysensor.init.open() from a Python shell as root it succeeds, so I think that means it's paired. If I open permissions on the USB device and do that as a normal user it complains `Signature verification failed. This device was probably paired with another computer.`

himekifee, 2021-01-09:
Not the same pairing. Ur one is SSL pairing which is as what you said but there is another pairing which do reset factory, flash firmware and calibrate. That is done by using validity tools, could you try that first please.

dlarrick, 2021-01-09:
Are you referring to the commands under "Error situations" section in the main README?
```
$ sudo systemctl stop python3-validity
$ sudo validity-sensors-firmware
$ sudo python3 /usr/share/python-validity/playground/factory-reset.py
```
Yes, I did that. Or is there some other command?

himekifee, 2021-01-09:
You need to wait for uunnicorn to respond then. I'm not sure about the code. You can try to do that several times though. In addition, 0097 and 0090 have the same size. I replaced my 0090 with 0097. 0097 is match on chip so faster and more accurate.

uunicorn, 2021-01-10:
Hi, 

0090 is a match-on-host device, as far as I know. 
`python-validity` was never designed to support this mode of operation. It is currently only supports enrolling/identifying on match-on-sensor devices. 
It does however support the pairing and flashing of 0090 devices - this feature was used by a @3v1n0 's pairing tool from his `fprintd` branch.

dlarrick, 2021-01-10:
Ahh, now it makes sense. Yes, 0090 is a match-on-host device, or at least nobody has ever figured out how to make it do match-on-device. Thanks. I'm closing this issue.

closed 2021-01-10 by dlarrick

scarlion1, 2025-01-13:
Well, it not make sense to me yet 🤪
What's solution nowadays to get 0090 working?  Where do we do enrolling/identifying if not with _python-validity_? (Debian 13 trixie/sid here)