uunicorn/python-validity #72: 138a:0094 Validity Sensors, Inc support?

parkerlreed, 2021-01-10:
```
Bus 001 Device 002: ID 138a:0094 Validity Sensors, Inc. 
```

I see this isn't mentioned in the udev or in the code. Is there a possiblity of this being supported?

uunicorn, 2021-01-10:
Some devices are reported to be very similar, but I know nothing about this device in particular. 
Is it a match-on-chip device?

parkerlreed, 2021-01-11:
From what I can recall from Validity90 yeah. Fingerprint had to be registered in Windows beforehand (Validity90 never got anywhere with 0094 as well)

uunicorn, 2021-01-12:
Then it is probably worth a shot. `python-validity` only supports match-on-chip devices at the moment. 0094 is not one of them, but if you have heaps of time, you can try setting up https://github.com/uunicorn/synaWudfBioUsb-sandbox project to extract necessary blobs.

Neonunux, 2026-09-28:
Hello,

I have this type of fingerprint reader, and I reverse-engineered it in order to implement a 100% free and open-source version.

The code and documentation are available here:
https://github.com/Neonunux/yoga-910-13ikb-80vf/blob/main/fingerprint/README.md

Although I would still like to test this driver further before submitting a pull request, it is working very well so far.