uunicorn/python-validity #74: Has anyone got Synaptics 06cb:0088 working with this?

jaxjexjox, 2021-01-12:
I would love to test it but due to the current release file issue, for the repository, I can't test.   I see several people getting other synaptic products working, but I have the 0088 derivant of the product.

Should I wait / hassle to get this release file fixed for the repo, or simply not even bother trying?

Bus 003 Device 005: ID 06cb:0088 Synaptics, Inc

uunicorn, 2021-01-12:
0088 is not supported at the moment as it appears to be a match-on-host device. 
See #34 for details.

vibbix, 2026-10-08:
@visorcraft any plans on upstreaming it? or just keeping it as an external patch?

visorcraft, 2026-10-31:
Hi @vibbix, no plans to upstream this. python-validity is unsupported at this point, so it'll stay as an external patch.