uunicorn/python-validity #75: [138a:0090] Delay since fprintd 1.90.4

MartinX3, 2021-01-20:
The fprintd project redirected me to this project.
https://gitlab.freedesktop.org/libfprint/fprintd/-/issues/102

With fprintd 1.90.1 everything is fine.

Now every time I need use my finger print for root stuff, I need to wait  
2 seconds in the terminal  
5 seconds in the GUI (pkexec)  
5 seconds in the login screen GUI  

```
(fprintd:85521): fprintd-DEBUG: 11:45:15.867: About to load configuration file '/etc/fprintd.conf'
(fprintd:85521): fprintd-DEBUG: 11:45:15.867: Launching FprintObject
(fprintd:85521): libfprint-context-DEBUG: 11:45:15.871: No driver found for USB device 17EF:1010
(fprintd:85521): libfprint-context-DEBUG: 11:45:15.871: No driver found for USB device 1D6B:0003
(fprintd:85521): libfprint-context-DEBUG: 11:45:15.871: No driver found for USB device 058F:9540
(fprintd:85521): libfprint-context-DEBUG: 11:45:15.871: No driver found for USB device 04F2:B541
(fprintd:85521): libfprint-context-DEBUG: 11:45:15.871: No driver found for USB device 8087:0A2B
(fprintd:85521): libfprint-vfs009x-DEBUG: 11:45:15.967: Initializing device
libusb: error [udev_hotplug_event] ignoring udev action change
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:16.222: [vfs0090] PROBE_STATE_LAST entering state 0
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:16.223: [vfs0090] PROBE_STATE_LAST entering state 1
(fprintd:85521): libfprint-context-DEBUG: 11:45:16.223: No driver found for USB device 046D:C05A
(fprintd:85521): libfprint-context-DEBUG: 11:45:16.223: No driver found for USB device 17EF:100F
(fprintd:85521): libfprint-context-DEBUG: 11:45:16.223: No driver found for USB device 17EF:1010
(fprintd:85521): libfprint-context-DEBUG: 11:45:16.223: No driver found for USB device 1D6B:0002
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:16.223: [vfs0090] PROBE_STATE_LAST entering state 2
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:16.223: [vfs0090] PROBE_STATE_LAST completed successfully
libusb: error [udev_hotplug_event] ignoring udev action change
(fprintd:85521): libfprint-device-DEBUG: 11:45:16.224: Device reported probe completion
(fprintd:85521): libfprint-device-DEBUG: 11:45:16.224: Completing action 1 in idle!
(fprintd:85521): fprintd-DEBUG: 11:45:16.229: entering main loop
(fprintd:85521): fprintd-DEBUG: 11:45:16.229: D-Bus service launched with name: net.reactivated.Fprint

(fprintd:85521): fprintd-DEBUG: 11:45:24.608: Requesting device 'Validity VFS0090' authorization for method ListEnrolledFingers from :1.325
(fprintd:85521): fprintd-DEBUG: 11:45:24.609: Getting authorization to perform Polkit action net.reactivated.fprint.device.setusername
(fprintd:85521): fprintd-DEBUG: 11:45:24.614: Getting authorization to perform Polkit action net.reactivated.fprint.device.verify
(fprintd:85521): fprintd-DEBUG: 11:45:24.617: Authorization granted to Validity VFS0090 to call method 'ListEnrolledFingers' for device :1.325!
(fprintd:85521): fprintd-DEBUG: 11:45:24.618: file_storage_discover_prints() for user 'martin' in '/var/lib/fprint/martin/vfs0090/0'
(fprintd:85521): fprintd-DEBUG: 11:45:24.620: Requesting device 'Validity VFS0090' authorization for method Claim from :1.325
(fprintd:85521): fprintd-DEBUG: 11:45:24.620: Getting authorization to perform Polkit action net.reactivated.fprint.device.setusername
(fprintd:85521): fprintd-DEBUG: 11:45:24.624: Getting authorization to perform Polkit action net.reactivated.fprint.device.verify
(fprintd:85521): fprintd-DEBUG: 11:45:24.627: Authorization granted to Validity VFS0090 to call method 'Claim' for device :1.325!
(fprintd:85521): fprintd-DEBUG: 11:45:24.627: user 'martin' claiming the device: 0
(fprintd:85521): libfprint-vfs009x-DEBUG: 11:45:24.724: Initializing device
libusb: error [udev_hotplug_event] ignoring udev action change
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:24.973: [vfs0090] INIT_STATE_LAST entering state 0
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:24.973: [vfs0090] INIT_STATE_LAST entering state 1
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:24.973: [vfs0090] INIT_STATE_LAST entering state 2
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:24.974: [vfs0090] INIT_STATE_LAST entering state 3
libusb: error [udev_hotplug_event] ignoring udev action change
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.133: [vfs0090] INIT_STATE_LAST entering state 4
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.146: [vfs0090] INIT_STATE_LAST entering state 5
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.147: [vfs0090] INIT_STATE_LAST entering state 6
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.152: [vfs0090] INIT_STATE_LAST entering state 7
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.152: [vfs0090] INIT_STATE_LAST entering state 8
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.152: [vfs0090] INIT_STATE_LAST entering state 9
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.152: [vfs0090] INIT_STATE_LAST entering state 10
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.152: [vfs0090] INIT_STATE_LAST entering state 11
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.152: [vfs0090] TLS_HANDSHAKE_STATE_LAST entering state 0
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.155: [vfs0090] TLS_HANDSHAKE_STATE_LAST entering state 1
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.157: [vfs0090] TLS_HANDSHAKE_STATE_LAST entering state 2
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.159: [vfs0090] TLS_HANDSHAKE_STATE_LAST entering state 3
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.463: [vfs0090] TLS_HANDSHAKE_STATE_LAST entering state 4
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.463: [vfs0090] TLS_HANDSHAKE_STATE_LAST completed successfully
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:25.463: [vfs0090] INIT_STATE_LAST completed successfully
(fprintd:85521): libfprint-device-DEBUG: 11:45:25.463: Device reported open completion
(fprintd:85521): libfprint-device-DEBUG: 11:45:25.463: Completing action 2 in idle!
(fprintd:85521): fprintd-DEBUG: 11:45:25.463: claimed device 0
(fprintd:85521): fprintd-DEBUG: 11:45:25.467: Requesting device 'Validity VFS0090' authorization for method VerifyStart from :1.325
[... 53 lines cut ...]
(fprintd:85521): libfprint-vfs009x-DEBUG: 11:45:27.120: USB read transfer cancelled

(fprintd:85521): libfprint-vfs009x-CRITICAL **: 11:45:27.120: LED blinking failed with error Vorgang wurde abgebrochen
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:27.120: [vfs0090] IMAGE_DOWNLOAD_STATE_LAST entering state 6
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:27.120: [vfs0090] IMAGE_DOWNLOAD_STATE_LAST entering state 9
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:27.120: [vfs0090] IMAGE_DOWNLOAD_STATE_LAST completed successfully
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:27.120: [vfs0090] SCAN_STATE_LAST completed successfully

(fprintd:85521): libfprint-vfs009x-CRITICAL **: 11:45:30.120: USB write transfer error: transfer timed out

(fprintd:85521): libfprint-vfs009x-CRITICAL **: 11:45:30.120: Data exchange failed at state 1, usb error: transfer timed out
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:30.120: [vfs0090] SSM DEACTIVATE_STATE_LAST failed in state 1 with error: transfer timed out
(fprintd:85521): libfprint-SSM-DEBUG: 11:45:30.120: [vfs0090] DEACTIVATE_STATE_LAST completed with error: transfer timed out

(fprintd:85521): libfprint-vfs009x-CRITICAL **: 11:45:30.120: Deactivation failed at state 1, unexpected device reply during deactivation
(fprintd:85521): libfprint-device-DEBUG: 11:45:30.120: Device reported generic error during action; action was: 6
(fprintd:85521): libfprint-device-DEBUG: 11:45:30.120: Device reported identify completion
(fprintd:85521): libfprint-device-DEBUG: 11:45:30.120: Completing action 6 in idle!
(fprintd:85521): fprintd-DEBUG: 11:45:30.120: identify_cb: result verify-no-match
(fprintd:85521): fprintd-DEBUG: 11:45:30.122: Requesting device 'Validity VFS0090' authorization for method Release from :1.325
(fprintd:85521): fprintd-DEBUG: 11:45:30.122: Authorization granted to Validity VFS0090 to call method 'Release' for device :1.325!
(fprintd:85521): libfprint-device-DEBUG: 11:45:30.123: Device reported close completion
(fprintd:85521): libfprint-device-DEBUG: 11:45:30.123: Completing action 3 in idle!
(fprintd:85521): fprintd-DEBUG: 11:45:30.123: released device 0
```

himekifee, 2021-01-20:
Well, that's still a `libfprint` issue. `python-validity` does not really work on MoH devices.

MartinX3, 2021-01-20:
Aaaargh. Wrong project.
Must be drunk.
Will move the issue to https://github.com/3v1n0/libfprint

closed 2021-01-20 by MartinX3