uunicorn/python-validity #76: Anyone working on 06cb:009b?

JRuedas, 2021-01-28:
I've seen that the 06cb:009a is already supported. This fingerprint sensor is used by the MSI Modern 14 and MSI Prestige 15. And probably more MSI models.

th3raid0r, 2021-03-14:
I'm also curious and willing to start trying to pull firmware blobs. Currently trying to learn it all.

johnhoth, 2021-03-21:
Well, I doesn't have deep understanding of distinctions in 06cb:009a and 06cb:009b. Why one is supported and the other is not? 

p.s. i have 06cb:009b

himekifee, 2021-03-21:
I guess no one has free time.

goldmont, 2021-04-22:
MSI Prestige 15 it's using 06cb:009b model. I need it too, please

hippo-o-matic, 2021-05-20:
I've managed to get https://github.com/uunicorn/synaWudfBioUsb-sandbox to at least recognize the device, and output something other than an error. Unfortunately, the output stops at `USB >>>>>>>>>> In DllMain reason=3` and doesn't respond to interaction with the sensor. I've attached the log from running `./run.sh enroll` in case it has something useful, but unfortunately I'm a bit lost at what to do next. 
[enroll.log](https://github.com/uunicorn/python-validity/files/6512954/enroll.log)
I'll keep poking at this tomorrow, but if anyone has any advice it would be appreciated.

goldmont, 2022-03-04:
Hi all... Any news about this?

ArsenShkryabin, 2026-11-03:
Hi everyone!

After years of this sensor (`06cb:009b`, MSI SecurePad / Modern / Prestige / Creator / GS66) being marked as unsupported, I've managed to build and verify a working Linux driver and drop-in `fprintd` D-Bus service:

👉 **https://github.com/ArsenShkryabin/synaptics-009b-fprintd**

### Key findings & architecture:
1. **EP 0x82 Stream Frames-Sync (Die Type-2):** Unlike older ThinkPad sensors (`009a`, `0097`) that poll via opcode `0x51`, the `009b` sensor is Die Type-2 and strictly requires the **EP 0x82 stream frames-sync** transport path. Attempting to send polled identify commands causes hardware faults and resets the sensor.
2. **On-Host Matching (MoH):** The hardware doesn't do match-on-chip; we perform on-host neural minutiae extraction and verification (QuantumMatcher).
3. **USB Stability:** Fixed the aggressive USB port reset loops present in earlier experimental code (`Soft` stale state policy + `open_any`), allowing clean device acquisition in <10ms with zero kernel xHCI churn.
4. **Automated Setup:** The installer pulls the official 1.2 MB driver package directly from Microsoft Update Catalog to carve model weights on Linux without requiring a Windows partition.

Fully functional with standard `fprintd-enroll`, `fprintd-verify`, and PAM authentication (`sudo`, screen lock). Feel free to test and star!

ImanolGo, 2026-11-16:
I've been working on the 06cb:009b in an MSI Prestige 15 A10SC (sensor type 0x2449, ROM 6.20, product 0x33, cf. #282) and have it working with python-validity 0.15 + open-fprintd: pairing, calibration, enrol/verify, lock screen and sudo.

Repo (patch for python-validity 0.15, tools, full write-up): https://github.com/ImanolGo/synaptics-009b-linux

Everything below comes from the Windows driver itself (`synaWudfBioUsb.dll` 5.5.5.1086): usbmon captures plus plaintext logs of its command path (Frida hook in a KVM guest), checked byte-for-byte where possible.

- **No flash.** `3e` returns JEDEC `ff:ff` / size 0, which is why python-validity stops at "Unknown flash IC". Windows never uploads a fwext or writes flash.
- **Pairing** is python-validity's sequence without flash: `01`, `19`, a 5 797-byte `06 02 00 00 01…` blob (from the DLL), `4f` with only the host cert, `01`, `50`, `1a`, TLS. The host keeps the `make_tls_flash()` blob (Windows: `Device Data\PairingData`). The 5.5.x driver wraps the host key with `password_hardcoded` directly (no DMI/GWK identity), so a Linux pairing can be converted for Windows and both can share it.
- The `50` blob is signed with the **ROM 6.20** firmware key from the DLL's per-ROM key table (python-validity's key is the 6.7 entry).
- **Match-on-host**: Windows sends no template, match or DB commands. Session: `1a 01 19 75`, then `06` + a 10 500-byte constant blob from the DLL. Capture: `02` → EP 0x83 `03 xx 04` done (`03 xx 06/07` = aborted) → `51 00200000` returns a 144 × 56 image over TLS.
- **Calibration and the capture command can be generated**: start-up blob, two calibration captures (frames on EP 0x82), corrections = python-validity's `average()`/`scale()` with factor 5/2, key line = correction rows 18/28/38 per third of the line (5 → 4), `0x43` from the sensor-type blob + bit-packed factory tag 3, `patch_timeslot_table(…, True, 2)`. The builder reproduces all eight `02` commands I recorded from Windows byte-for-byte. Only chunk `0x48` (clean slate) is approximated, and capture also works without it.
- Matching: SIFT + RANSAC (OpenCV) on the 144 × 56 images; genuine 16–89 inliers vs impostor ≤ 4 on my (small) dataset.

@ArsenShkryabin interesting that your implementation streams on EP 0x82 and found `0x51` problematic. On my unit the Windows driver reads every image with `51 00200000` over TLS (EP 0x82 only carries the two calibration frames), and that works from Linux as well. Maybe different firmware or driver versions; if you can share your `01`/`75` output and the driver version you extract from, it would be good to compare.

I only have one device. Output of `sudo bash tools/run_9b.sh` (`75`, `01`, factory bits) and the calibration error value from other 009b owners would help confirm what is per-device.