uunicorn/python-validity #79: 06cb:009a verify-no-match after enroll-completed

HankvanHinton, 2021-02-08:
Hello everyone, 

on my Thinkpad T480s running Manjaro KDE (Kernel 5.9.16), I can enroll my finger but verification then always fails. I've tried resetting and re-enrolling my finger multiple times. 

`Using device /net/reactivated/Fprint/Device/0
Enrolling right-index-finger finger.
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-stage-passed
Enroll result: enroll-completed
`
`Using device /net/reactivated/Fprint/Device/0
Listing enrolled fingers:
 - #0: WINBIO_ANSI_381_POS_RH_INDEX_FINGER
Verify result: verify-no-match (done)
`

Please let me know if I need to supply more information.

KoalaV2, 2021-03-06:
+1 Same issue here. T480s and arch linux.

KoalaV2, 2021-03-27:
Any updates on this?

blondiejon, 2021-08-11:
+1 Also for me. T480s and Elementary OS 6 Odin.

biladina, 2021-09-08:
+1 its same here, using P52 with Fedora 34..

heeen, 2021-09-10:
I had the same issue on a T480, I did the reset procedure and rebooted that seemed to fix it

uunicorn, 2021-09-26:
A factory reset forces a calibration. If you put your finger/palm on the sensor during the calibration, all subsequent attempts to enroll/verify will be very unreliable. `06cb:009a` device is special in a sense that calibration results are partly stored on both the device and the host. Whenever you calibrate the sensor, remember to delete `/usr/share/python-validity/calib-data.bin`.

closed 2021-09-26 by uunicorn

KoalaV2, 2021-09-26:
Can confirm factory reset fixed it for me, thanks!

biladina, 2021-10-01:
beside factory reset, try to clear fingerprint data from BIOS also fix my problem..