uunicorn/python-validity #80: Use without systemd

void-m4110c, 2021-02-20:
Hi, 

as there are many distros out there that don't force you to use systemd, it would be great to have an option to install python-validity on those systems. 

Is there any solution for this?

Thanks

himekifee, 2021-02-20:
Look at the systemd service file and write your own script.

mid-kid, 2021-03-02:
I'm running this on gentoo+OpenRC just fine. The init script looks like:
```
$ cat /etc/init.d/validity
#!/sbin/openrc-run
command="/usr/lib/python-validity/dbus-service"
command_background=true
pidfile="/var/run/${RC_SVCNAME}.pid"
```

Midou36O, 2022-08-17:
> I'm running this on gentoo+OpenRC just fine. The init script looks like:
> 
> ```
> $ cat /etc/init.d/validity
> #!/sbin/openrc-run
> command="/usr/lib/python-validity/dbus-service"
> command_background=true
> pidfile="/var/run/${RC_SVCNAME}.pid"
> ```

@mid-kid without fprintd-clients? I'm struggling to install it as it has a hard requirement for libsystemd

mid-kid, 2022-08-19:
@Midou36O You can use fprintd-0.9.0 to build the clients. Here's a relevant patch for it: https://pastebin.com/raw/7AQbB0WC

Midou36O, 2022-08-19:
So i could just use a modern libelogind patched fprintd and it would just work?

mid-kid, 2022-08-21:
Like I said, I use the clients from fprintd-0.9.0, which don't need libsystemd, so I wouldn't know.

Midou36O, 2022-08-22:
I see, thanks for the help.

Midou36O, 2023-08-17:
Update: I got it to work, [Please check this if you want to install it through OpenRC](https://github.com/uunicorn/python-validity/issues/67#issuecomment-1682346902)