uunicorn/python-validity #87: Exception: Unexpected TLS version 4 0

yunniel, 2021-06-23:
I'm trying to use your code on a P51 with 138a:0097 sensor.
And the same  exception as https://github.com/uunicorn/python-validity/issues/55 just jumped out when I tried to initialize the device, as below:

`Found device <DEVICE ID 138a:0097 on Bus 001 Address 005>
Downloading https://download.lenovo.com/pccbbs/mobiles/n1mgf03w.exe to extract 6_07f_lenovo_mis.xpfwext
Extracting "Synaptics Metallica MOC Touch Fingerprint Reader Driver for Windows 8.1 64-bit" - setup data version 5.5.0
 - "app/WBF_Drivers/6_07f_lenovo_mis.xpfwext"
Done.
Found firmware at /tmp/tmp_50x3jwx/app/WBF_Drivers/6_07f_lenovo_mis.xpfwext
The device will be now reset to factory and associated to the current laptop.
Press Enter to continue (or Ctrl+C to cancel)...
Opening device 0x97
Factory reset...
Sleeping...
Pairing the sensor with device 20HJS0AR2P
Opening device 0x97
Initializing flash...
Detected Flash IC: W25Q80B, 1048576 bytes
Sleeping...
Try 1 failed with error: Failed: 04af
Opening device 0x97
Initializing flash...
Sleeping...
Try 2 failed with error: Unexpected TLS version 4 0
Opening device 0x97
Initializing flash...
Sleeping...
Try 3 failed with error: Unexpected TLS version 4 0
Opening device 0x97
Initializing flash...
Sleeping...
Try 4 failed with error: Unexpected TLS version 4 0
Opening device 0x97
Initializing flash...
Sleeping...
Try 5 failed with error: Unexpected TLS version 4 0
Device didn't reply in time...
Traceback (most recent call last):
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 339, in <module>
    vfs_tools.initialize(fwpath, args.calibration_data)
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 249, in initialize
    self.pair(fwpath, calib_data)
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 220, in pair
    self.retry_command(init_flash_command, max_retries=5)
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 112, in retry_command
    raise(err)
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 103, in retry_command
    command()
  File "/snap/validity-sensors-tools/65/vfs-tools/validity-sensors-tools.py", line 219, in init_flash_command
    init_flash()
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/init_flash.py", line 97, in init_flash
    partition_flash(flash_layout_hardcoded, client_public)
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/init_flash.py", line 70, in partition_flash
    info = get_flash_info()
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/flash.py", line 28, in get_flash_info
    rsp=tls.cmd(unhex('3e'))
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/tls.py", line 104, in cmd
    rsp=self.app(cmd)
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/tls.py", line 138, in app
    return self.parse_tls_response(self.usb.cmd(self.make_app_data(b)))
  File "/snap/validity-sensors-tools/65/vfs-tools/proto9x/tls.py", line 336, in parse_tls_response
    raise Exception('Unexpected TLS version %d %d' % (mj, mn))
Exception: Unexpected TLS version 4 0
`

I tried the same solution given there and it didn't worked. Instead when using the correction given in the README section

`sudo systemctl status python3-validity
`
The following message comes up

`python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/lib/systemd/system/python3-validity.service; static; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2021-06-23 16:27:30 CEST; 26min ago
    Process: 28828 ExecStart=/usr/lib/python-validity/dbus-service --debug (code=exited, status=1/FAILURE)
   Main PID: 28828 (code=exited, status=1/FAILURE)

juin 23 16:27:30 user0 systemd[1]: Started python-validity driver dbus service.
juin 23 16:27:30 user0 dbus-service[28828]: Traceback (most recent call last):
juin 23 16:27:30 user0 dbus-service[28828]:   File "/usr/lib/python-validity/dbus-service", line 17, in <module>
juin 23 16:27:30 user0 dbus-service[28828]:     import dbus
juin 23 16:27:30 user0 dbus-service[28828]: ModuleNotFoundError: No module named 'dbus'
juin 23 16:27:30 user0 systemd[1]: python3-validity.service: Main process exited, code=exited, status=1/FAILURE
juin 23 16:27:30 user0 systemd[1]: python3-validity.service: Failed with result 'exit-code'.
`

uunicorn, 2021-06-23:
It looks like your are using a mixture of tools here.  You don't need `validity-sensors-tools` to run `python-validity`.

yunniel, 2021-06-24:
Actually, I removed `validity-sensors-tools` and I used the commands below to set up

```
$ sudo apt remove fprintd
$ sudo add-apt-repository ppa:uunicorn/open-fprintd
$ sudo apt-get update
$ sudo apt install open-fprintd fprintd-clients python3-validity
$ fprintd-enroll
```
but I get a ` list_devices failed: ` message. So, I use the correction you preconize, and I tried to check the logs of the `python3-validity` daemon using `$ sudo systemctl status python3-validity` but I obtained the following error

```
● python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/lib/systemd/system/python3-validity.service; static; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2021-06-24 10:45:21 CEST; 5s ago
    Process: 198617 ExecStart=/usr/lib/python-validity/dbus-service --debug (code=exited, status=1/FAILURE)
   Main PID: 198617 (code=exited, status=1/FAILURE)

juin 24 10:45:21 user0 systemd[1]: Started python-validity driver dbus service.
juin 24 10:45:21 user0 dbus-service[198617]: Traceback (most recent call last):
juin 24 10:45:21 user0 dbus-service[198617]:   File "/usr/lib/python-validity/dbus-service", line 17, in <module>
juin 24 10:45:21 user0 dbus-service[198617]:     import dbus
juin 24 10:45:21 user0 dbus-service[198617]: ModuleNotFoundError: No module named 'dbus'
juin 24 10:45:21 user0 systemd[1]: python3-validity.service: Main process exited, code=exited, status=1/FAILURE
juin 24 10:45:21 user0 systemd[1]: python3-validity.service: Failed with result 'exit-code'.
```

the `start` option does nothing and the `enable` option throws :+1: 

```
sudo systemctl enable python3-validity
The unit files have no installation config (WantedBy=, RequiredBy=, Also=,
Alias= settings in the [Install] section, and DefaultInstance= for template
units). This means they are not meant to be enabled using systemctl.
 
Possible reasons for having this kind of units are:
• A unit may be statically enabled by being symlinked from another unit's
  .wants/ or .requires/ directory.
• A unit's purpose may be to act as a helper for some other unit which has
  a requirement dependency on it.
• A unit may be started when needed via activation (socket, path, timer,
  D-Bus, udev, scripted systemctl call, ...).
• In case of template units, the unit is meant to be enabled with some
  instance name specified.
```

going further, I tried a factory reset of the device using 

```
$ sudo systemctl stop python3-validity
$ sudo validity-sensors-firmware
```
and I obtained

```
Downloading https://download.lenovo.com/pccbbs/mobiles/nz3gf07w.exe to extract 6_07f_lenovo_mis_qm.xpfwext
Extracting "version 5.3.3542.26 (NZ3GF07W)" - setup data version 5.5.7 (unicode)
 - "codeGetExtractPath/WBF_Drivers/6_07f_lenovo_mis_qm.xpfwext"
Done.
Found firmware at /tmp/tmp4j0jsnru/codeGetExtractPath/WBF_Drivers/6_07f_lenovo_mis_qm.xpfwext
```
but the next step `$ sudo python3 /usr/share/python-validity/playground/factory-reset.py` also rises an error, as follows

```
Traceback (most recent call last):
  File "/usr/share/python-validity/playground/factory-reset.py", line 1, in <module>
    from validitysensor.sensor import factory_reset, RebootException
ModuleNotFoundError: No module named 'validitysensor'
```

This is the whole story , Thanks in advance for your support

uunicorn, 2021-06-24:
This is odd. The root cause seems to be this:

> juin 24 10:45:21 user0 dbus-service[198617]: ModuleNotFoundError: No module named 'dbus'

It does not look like you have dbus python bindings installed.
This is very strange because `python3-validity` package from PPA must explicitly depend on `python3-dbus`.

yunniel, 2021-06-25:
Any idea on how to solve this issue?

> juin 24 10:45:21 user0 dbus-service[198617]: ModuleNotFoundError: No module named 'dbus'

luciorq, 2021-08-12:
Hi @yunniel, you are using Ubuntu, right?

The root of the problem looks like you are using a not default install of Python, which is not searching for the modules where they were installed by the PPA.

You can check where the `validitysensor` module is currently installed using:

```shell
find /usr/lib -type d -iname 'validitysensor'
```

And you can check where your Python install is searching for paths using:

```shell
python3 -c 'import sys; print(*sys.path, sep = "\n")'
```

There are several ways of modifying your python installation to be forced to search modules at this path.
But I think a more stable solution (at least will not modify your python installations), is to modify the systemd service file to include the PYTHONPATH variable.

- Add this line `Environment=PYTHONPATH=$PYTHONPATH:/usr/lib/python3/dist-packages
` to `/lib/systemd/system/python3-validity.service` after `[Service]` header

The file should look something like this:

```ini
[Unit]
Description=python-validity driver dbus service

[Service]
Type=simple
ExecStart=/usr/lib/python-validity/dbus-service --debug
Restart=no

## Start of luciorq edit
Environment=PYTHONPATH=$PYTHONPATH:/usr/lib/python3/dist-packages
## End of luciorq edit
```

After modifying the file you need to reload the systemd daemon and restart the service

```shell
systemctl status python3-validity.service
```

sniffer, 2022-03-07:
I was getting this error with https://github.com/3v1n0/python-validity (validity-sensors-tools) project which it seems is now obsolete.

Have a look at https://gitter.im/Validity90/Lobby?at=601087ce063b6c68d51cb3bb uunicorn comment.

Maybe someone finds it usefull.

mjzjz, 2023-05-02:
> Hi @yunniel, you are using Ubuntu, right?嗨 @yunniel ，您使用的是 Ubuntu，对吗？
> 
> The root of the problem looks like you are using a not default install of Python, which is not searching for the modules where they were installed by the PPA.问题的根源看起来像您使用的不是默认安装的 Python，它没有搜索 PPA 安装它们的模块。
> 
> You can check where the `validitysensor` module is currently installed using:您可以使用以下命令检查当前安装 `validitysensor` 模块的位置：
> 
> ```shell
> find /usr/lib -type d -iname 'validitysensor'
> ```
> 
> And you can check where your Python install is searching for paths using:您可以使用以下命令检查您的 Python 安装在哪里搜索路径：
> 
> ```shell
> python3 -c 'import sys; print(*sys.path, sep = "\n")'
> ```
> 
> There are several ways of modifying your python installation to be forced to search modules at this path.有几种方法可以修改 python 安装以强制在此路径中搜索模块。 But I think a more stable solution (at least will not modify your python installations), is to modify the systemd service file to include the PYTHONPATH variable.但我认为更稳定的解决方案（至少不会修改您的 python 安装）是修改 systemd 服务文件以包含 PYTHONPATH 变量。
> 
> * Add this line `Environment=PYTHONPATH=$PYTHONPATH:/usr/lib/python3/dist-packages ` to `/lib/systemd/system/python3-validity.service` after `[Service]` header在 `[Service]` 标头之后添加此行 `Environment=PYTHONPATH=$PYTHONPATH:/usr/lib/python3/dist-packages ` 到 `/lib/systemd/system/python3-validity.service`
> 
> The file should look something like this:该文件应如下所示：
> 
> ```ini
> [Unit]
> Description=python-validity driver dbus service
> 
> [Service]
> Type=simple
> ExecStart=/usr/lib/python-validity/dbus-service --debug
> Restart=no
> 
> ## Start of luciorq edit
> Environment=PYTHONPATH=$PYTHONPATH:/usr/lib/python3/dist-packages
> ## End of luciorq edit
> ```
> 
> After modifying the file you need to reload the systemd daemon and restart the service修改文件后需要重新加载systemd守护进程并重启服务
> 
> ```shell
> systemctl daemon-reload
> 
> systemctl restart python3-validity.service
> 
> systemctl status python3-validity.service
> ```
> 
> This solution is not permanent, if the file get reinstalled from the PPA, the modifications will be overwritten.此解决方案不是永久性的，如果从 PPA 重新安装文件，修改将被覆盖。

this works for me
i use manjaro and use anaconda.
[image]
apparently there are problems about the environment path.
i just try your steps and it works.
thank you

rkochar, 2024-04-09:
Doesn't work anymore on Manjaro.

```
> systemctl status python3-validity.service
× python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/usr/lib/systemd/system/python3-validity.service; disabled; preset: disabled)
     Active: failed (Result: exit-code) since Tue 2024-04-09 21:11:24 CEST; 5s ago
   Duration: 86ms
    Process: 80688 ExecStart=/usr/lib/python-validity/dbus-service --debug (code=exited, status=1/FAILURE)
   Main PID: 80688 (code=exited, status=1/FAILURE)
        CPU: 86ms

Apr 09 21:11:24 rahul-g5 systemd[1]: Started python-validity driver dbus service.
Apr 09 21:11:24 rahul-g5 dbus-service[80688]: Traceback (most recent call last):
Apr 09 21:11:24 rahul-g5 dbus-service[80688]:   File "/usr/lib/python-validity/dbus-service", line 24, in>
Apr 09 21:11:24 rahul-g5 dbus-service[80688]:     from validitysensor import init
Apr 09 21:11:24 rahul-g5 dbus-service[80688]: ModuleNotFoundError: No module named 'validitysensor'
Apr 09 21:11:24 rahul-g5 systemd[1]: python3-validity.service: Main process exited, code=exited, status=1>
Apr 09 21:11:24 rahul-g5 systemd[1]: python3-validity.service: Failed with result 'exit-code'.
```

```
> find /usr/lib -type d -iname 'validitysensor'
/usr/lib/python3.9/site-packages/validitysensor

> python3 -c 'import sys; print(*sys.path, sep ="\n")'
/home/rkochar/Library/validity-sensors-tools-git
/usr/lib/python3.9/site-packages
/home/rkochar/.pyenv/versions/3.9.16/lib/python39.zip
/home/rkochar/.pyenv/versions/3.9.16/lib/python3.9
/home/rkochar/.pyenv/versions/3.9.16/lib/python3.9/lib-dynload
/home/rkochar/.pyenv/versions/3.9.16/lib/python3.9/site-packages
```

JGSphaela, 2024-04-13:
@rkochar Hi, have you try to add the python path that contain the correct package to the systemctl service like @luciorq mentioned above? I'm also on Manjaro and it solve the problem.

I also want to add that it might be the case that the line also need to be add to `open-fprintd.service` because in my case both `open-fprintd.service` and `python3-validity.service` complain about the missing module.

Could you check that if adding `Environment=PYTHONPATH=/usr/lib/python3.9/site-packages/` to both `open-fprintd.service` and `python3-validity.service` systemctl file works?

rkochar, 2024-04-13:
Hi, @JGSphaela . The original solution indeed works. I had a typo in PYTHONPATH. I have deleted the comment.