uunicorn/python-validity #89: 06cb:009a can

treasuretron, 2021-07-04:


sudopluto, 2021-07-06:
https://github.com/uunicorn/python-validity/issues/54#issuecomment-742575505

this ended up working for me, maybe need to mess around with rpm pkgs?

ottotivadar, 2021-08-09:
Hello!
I had the same problem on Lenovo T470s and Fedora 34. This is caused by SELinux. SELinux prevent the services from creating the necessary files.

The above mentioned comment solved the issue, because the files created manually. To bypass the problem, put SELinux into permissive mode until restart:
sudo setenforce Permissive

... and restart the service:
sudo systemctl restart python3-validity

After this it should work.

On Fedora 34 I faced another issue after this: Fedora did not used the fingerprint for authentication. To solve it:
sudo authselect enable-feature with-fingerprint

(I just started to play with this package, so I reserve the right to make a mistake. :) )

silviuvulcan, 2021-11-17:
Same issue on fedora 35 and https://github.com/uunicorn/python-validity/issues/54#issuecomment-742575505 fixed it. Might be useful to link those to instructions.

haoyun, 2023-07-29:
What should I do if I would like to keep SELinux enforcing?

HatWearingMann, 2023-09-24:
> Hello! I had the same problem on Lenovo T470s and Fedora 34. This is caused by SELinux. SELinux prevent the services from creating the necessary files.
>
> The above mentioned comment solved the issue, [...]

omfg thank you so much

uwe-ss, 2024-11-18:
> Hello! I had the same problem on Lenovo T470s and Fedora 34. This is caused by SELinux. SELinux prevent the services from creating the necessary files.
>
> The above mentioned comment solved the issue, [...]

Maybe, there is a more elegant solution but I did not find any, so this one was the solution for me, too, on Lenovo X380 (with Validity ID 138a:009d) and Fedora 40.