uunicorn/python-validity #91: 138a:0090 is not working anymore on Ubuntu 20.04

kecsap, 2021-07-20:
I get the "Exception: Failed: 04af" error when the python3-validity.service starts. Maybe the binary blobs need an update?
There is a new driver at the Lenovo site:

https://download.lenovo.com/pccbbs/mobiles/n1cgn09w.exe

bookshiyi, 2021-12-04:
same issue, mark.

```
$ fprintd-enroll 
Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: No devices available
```

osbre, 2022-04-14:
Restarting it solved this problem:

```shell
sudo systemctl restart python3-validity
```

Nyamkhuub, 2022-04-22:
> Restarting it solved this problem:
> 
> ```shell
> sudo systemctl restart python3-validity
> ```

What is your laptop model?

osbre, 2022-04-22:
> > Restarting it solved this problem:
> > ```shell
> > sudo systemctl restart python3-validity
> > ```
> 
> What is your laptop model?

ThinkPad T470

ipstone, 2023-11-17:
hey @osbre , can you confirm it's working for you /fingerprint setup/function properly? Somewhere else I read, it seems that this finger-printer reader isn't supported: 
https://github.com/uunicorn/python-validity/issues/71

I am using a thinkpad carbon x1 gen4, with also 138a:0090, I am on ubuntu 22.04, I am encountering the same problem as listed in this issue. Perhaps it works on elementary OS?

thanks