uunicorn/python-validity #96: There is no support for Ubuntu 21.04

vzubovich, 2021-09-07:
I've got "The repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu hirsute Release' does not have a Release file."

vzubovich, 2021-09-18:
Many thanks for your advice, but it seems it doesn't work for me. Now I'm getting an output like this:
```
W: Skipping acquire of configured file 'InRelease/binary-amd64/Packages' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/binary-i386/Packages' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/i18n/Translation-en_US' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/i18n/Translation-en' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/dep11/Components-amd64.yml' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/dep11/icons-48x48.tar' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/dep11/icons-64x64.tar' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/dep11/icons-64x64@2.tar' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/dep11/icons-128x128.tar' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'InRelease/cnf/Commands-amd64' as repository 'http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu groovy InRelease' doesn't have the component 'InRelease' (component misspelt in sources.list?)
```

uunicorn, 2021-09-26:
I've just published 0.13 to PPA for Focal, Hirsute, Impish & Bionic. 
Please reopen if it still does not work.

closed 2021-09-26 by uunicorn