uunicorn/python-validity #102: Support for 27c6:538d Shenzhen Goodix Technology Co.,Ltd. FingerPrint reader

litch0, 2021-11-07:
Hi there,
I have tried using this library with the `27c6:538d Shenzhen Goodix Technology Co.,Ltd. FingerPrint` reader, but when I started rebooted the system i got this error:
```sh
╰─○ sudo systemctl status python3-validity
× python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/usr/lib/systemd/system/python3-validity.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sun 2021-11-07 15:17:27 -03; 2min 19s ago
   Main PID: 1073 (code=exited, status=1/FAILURE)
        CPU: 105ms

Nov 07 15:17:27 fedora systemd[1]: python3-validity.service: Scheduled restart job, restart counter is at 5.
Nov 07 15:17:27 fedora systemd[1]: Stopped python-validity driver dbus service.
Nov 07 15:17:27 fedora systemd[1]: python3-validity.service: Start request repeated too quickly.
Nov 07 15:17:27 fedora systemd[1]: python3-validity.service: Failed with result 'exit-code'.
Nov 07 15:17:27 fedora systemd[1]: Failed to start python-validity driver dbus service.
```

and then when I tried again:
```sh
Nov 07 15:21:40 fedora dbus-service[4296]: /usr/lib/python-validity/dbus-service:33: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
Nov 07 15:21:40 fedora dbus-service[4296]:   GLib.threads_init()
Nov 07 15:21:40 fedora dbus-service[4296]: /usr/lib/python-validity/dbus-service:245: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load fo>
Nov 07 15:21:40 fedora dbus-service[4296]:   config = yaml.load(configfd)
Nov 07 15:21:40 fedora python3[4296]: detected unhandled Python exception in '/usr/lib/python-validity/dbus-service'
Nov 07 15:21:40 fedora python3[4296]: can't communicate with ABRT daemon, is it running? [Errno 13] Permission denied
Nov 07 15:21:40 fedora python3[4296]: error sending data to ABRT daemon: 
Nov 07 15:21:40 fedora dbus-service[4296]: Traceback (most recent call last):
Nov 07 15:21:40 fedora dbus-service[4296]:   File "/usr/lib/python-validity/dbus-service", line 307, in <module>
Nov 07 15:21:40 fedora dbus-service[4296]:     main()
Nov 07 15:21:40 fedora dbus-service[4296]:   File "/usr/lib/python-validity/dbus-service", line 260, in main
Nov 07 15:21:40 fedora dbus-service[4296]:     backoff()
Nov 07 15:21:40 fedora dbus-service[4296]:   File "/usr/lib/python-validity/dbus-service", line 214, in backoff
Nov 07 15:21:40 fedora dbus-service[4296]:     with open(backoff_file, 'w') as f:
Nov 07 15:21:40 fedora dbus-service[4296]: PermissionError: [Errno 13] Permission denied: '/usr/share/python-validity/backoff'
Nov 07 15:21:40 fedora systemd[1]: python3-validity.service: Main process exited, code=exited, status=1/FAILURE
```
the full journalctl is just a repetition of this message. Can anybody help?

JoaoMirandaXP, 2021-12-04:
Hi There, I've got a similar problem, but in my case the packages didn't even installed properly.

lmtr0, 2021-12-05:
Oh what a coincidence, I am on a dell Inspirion 5502

lmtr0, 2022-01-28:
the device encrypts the information before it's pass down to the driver, so If we don't know the decryption key, we can't really interact in a reliable way with the reader.
[image]
more about [this](https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/425)
however they are working on it
[image]
more about it on this [issue](https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/344)