uunicorn/python-validity #103: Lenovo X1 Yoga 2nd (138a:0097) + Fedora = various problems

FlyinPancake, 2021-11-08:
Hi all, 
I have tried installing the driver but I couldn't get it to work. What would be the correct procedure (I followed the instructions on from the README) ?

I'll provide the logs that are needed, I just don't know where to begin.

kambiz, 2021-11-11:
Hello, I too am having issues with Fedora 35 and a Lenovo ThinkPad T470 with a 138a:0097 Validity Sensor fingerprint device.I have gotten this to python-validity package to work with the same ThinkPad T470 running Ubuntu 21.10 so I am no sure what is wrong. 

In Fedora 35; I run: 
`$ sudo dnf copr enable tigro/python-validity`
Then:
`$ sudo dnf install open-fprintd fprintd-clients fprintd-clients-pam python3-validity`

It shows everything is installed, and complete. And upon running fprintd-enroll, I get this; 
`Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: `

What is going on and can anyone point me and the original poster to a fix? Like I said, I have had this package working in Ubuntu 21.10 with the same laptop but with Fedora I run into this error.

kambiz, 2021-11-11:
I have made some progress into this issue and it seems to be that some files are missing and ownership settings are incorrect. I got fprintd to enroll my finger print with the following: 

```
sudo touch /usr/share/python-validity/backoff
sudo touch /usr/share/python-validity/calib-data.bin
```

Then: 

```
cd /usr/share/python-validity/ && ls -la
sudo chmod 755 6_07f_lenovo_mis_qm.xpfwext
```

After this, I restarted the python-3-validity services in systemctl; rebooted my system and was able to enroll my index finger with `fprintd-enroll` and it is verified! Now, at this point however, I cannot get Gnome to allow me to use my fingerprint to unlock my account and Gnome Keyring.

arfejes, 2022-01-07:
Hi Guys,

I ran into the very same issue on a Yoga 370. The above described solution works perfectly, thanks a lot.

I've tested this on both latest Manjaro and Pop! 20.04 as well as on Pop! 21.10 and I did not experience any issues there.

FlyinPancake, 2022-01-07:
Well I tried the solution provided by @kambiz. When I try to `fprintd-enroll` i get the following
```
➜ fprintd-enroll
Using device /net/reactivated/Fprint/Device/4
Enrolling right-index-finger finger.
Enroll result: enroll-stage-passed
Enroll result: enroll-failed
EnrollStop failed: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Remote peer disconnected
```
The error occures when I lift my finger up from the fingerprint sensor.

FlyinPancake, 2022-01-07:
Well I tried the solution provided by @kambiz. When I try to `fprintd-enroll` i get the following
```
➜ fprintd-enroll
Using device /net/reactivated/Fprint/Device/4
Enrolling right-index-finger finger.
Enroll result: enroll-stage-passed
Enroll result: enroll-failed
EnrollStop failed: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Remote peer disconnected
```
The error occures when I lift my finger up from the fingerprint sensor

FlyinPancake, 2022-03-22:
I've tried with a clean install of fedora and still no success

np22-jpg, 2022-07-04:
I tried the solution @kambiz gave. After using `systemctl restart python3-validity.service` to restart the service before using `fprintd-enroll`, everything appeared to work. In my case, I also had to use `systemctl enable` to make it work at startup.

Thank you!

dreamer, 2023-04-02:
I had the same problem with Fedora 38 running on Thinkpad X1 Carbon 3rd Gen. Also, the existing copr repo does not work any more due to changed Python ABI and some other changes.

I forked @Tigro's copr, fixed it up and rebuilt for Fedora 38: https://copr.fedorainfracloud.org/coprs/pbo/python-validity/

To enable it:

    $ sudo dnf copr enable pbo/python-validity

All other instructions are the same. After applying manual fix by @kambiz https://github.com/uunicorn/python-validity/issues/103#issuecomment-966016605 the sensor now works correctly.

> I have made some progress into this issue and it seems to be that some files are missing and ownership settings are incorrect. I got fprintd to enroll my finger print with the following:
> 
> ```
> sudo touch /usr/share/python-validity/backoff
> sudo touch /usr/share/python-validity/calib-data.bin
> ```
> 
> Then:
> 
> ```
> cd /usr/share/python-validity/ && ls -la
> sudo chmod 755 6_07f_lenovo_mis_qm.xpfwext
> ```
> 
> After this, I restarted the python-3-validity services in systemctl; rebooted my system and was able to enroll my index finger with `fprintd-enroll` and it is verified! Now, at this point however, to get this to work with Gnome do the following:
> 
> ```
> $ sudo authselect current
> $ sudo authselect enable-feature with-fingerprint
> $ sudo authselect apply-changes
> ```

Tigro, 2023-04-03:
> I had the same problem with Fedora 38 running on Thinkpad X1 Carbon 3rd Gen. Also, the existing copr repo does not work any more due to changed Python ABI and some other changes.
> 
> I forked @Tigro's copr, fixed it up and rebuilt for Fedora 38: https://copr.fedorainfracloud.org/coprs/pbo/python-validity/
> 
> To enable it:
> 
> ```
> $ sudo dnf copr enable pbo/python-validity
> ```
> 
> All other instructions are the same. After applying manual fix by @kambiz [#103 (comment)](https://github.com/uunicorn/python-validity/issues/103#issuecomment-966016605) the sensor now works correctly.
> 
> > ```
> > sudo touch /usr/share/python-validity/backoff
> > sudo touch /usr/share/python-validity/calib-data.bin
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Then:
> > ```
> > cd /usr/share/python-validity/ && ls -la
> > sudo chmod 755 6_07f_lenovo_mis_qm.xpfwext
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > After this, I restarted the python-3-validity services in systemctl; rebooted my system and was able to enroll my index finger with `fprintd-enroll` and it is verified! (…)

Hi, dreamer

Can you send me patch, and I will rebuild packages?