uunicorn/python-validity #105: 04f3:0c4b Elan Microelectronics Corp. ELAN:Fingerprint Not Supported.

mostafa-yasen, 2021-12-07:
Hey, I just got a new Lenovo Thinkbook-15 with the following specs as an output from `uname -a` command.

`Linux thinkbook-15 5.11.0-41-generic #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
With a fingerprint sensor which seem not to be supported with most of the packages. `04f3:0c4b Elan Microelectronics`

I went through the guideline in the readme file, but, failed to find a supported device when I run `fprintd-enroll`.

I took a sneak peek to the file `prototype.py` and found that my laptop's sensor is not supported.

[image]

I get the following as an output to `lsusb` command

[image]

Is there anyone who can help me with this?

Please let me know if you need more info.
Thanks.

pms967, 2022-01-06:
> fingerprint sensor which seem not to be supported with most of the packages. `04f3:0c4b Elan Microelectronics`

same laptop & problem here. It depends by the fact that this device is not (yet?) supported by libfprint. :(

suheyl-code, 2022-02-13:
I've the same problem.
20SL Lenovo ThinkBook 14-IIL
Bus 003 Device 002: ID 04f3:0c4b Elan Microelectronics Corp. ELAN:Fingerprint

heavenly999, 2022-02-24:
Same here, would need a driver.

harshhx17, 2022-03-07:
Please update here if anyone finds a driver which supports this device in the future.

jackfido, 2022-03-13:
Same here:

[image]

nioliz, 2022-06-03:
Same problem, Ubuntu 22.04 LTS
Bus 003 Device 002: ID 04f3:0c4b Elan Microelectronics Corp. ELAN:Fingerprint

uunicorn, 2022-06-14:
Hi, this project was created exclusively for Validity (Synaptics) products. 
Unless it's some kind of OEM arrangement, I really doubt it will be ever supported by this project.

Xavier-Platinum, 2022-07-09:
The same issue as above please if there is any help anyone with an answer should help out, thank you

utbuiquoc, 2022-07-14:
same problem

LAfricain, 2022-07-14:
Hi,
I think [here](https://gitlab.freedesktop.org/libfprint/libfprint/-/issues) is a better place for a report. No? I add a comment in this [https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/214](issue)

jackfido, 2022-08-28:
> Hi, I think [here](https://gitlab.freedesktop.org/libfprint/libfprint/-/issues) is a better place for a report. No? I add a comment in this [issue](https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/214)

I follow that, and Im now able to see "working" my hardware, I use quotes because it works, it can capture my finger print... but it capture quick and it shows bad, then it causes an error...

I should to remove libfprintd as shows here: https://www.tec4tric.com/linux/enable-fingerprint-login-in-ubuntu and then install it again to see my device work with fprintd-enroll command
RESULT:

List my device:
`goliapp@goliapp-ubuntu:~$ fprintd-list goliapp
found 1 devices
Device at /net/reactivated/Fprint/Device/0
Using device /net/reactivated/Fprint/Device/0
User goliapp has no fingers enrolled for ElanTech Fingerprint Sensor.
goliapp@goliapp-ubuntu:~$`

Actual error:
`goliapp@goliapp-ubuntu:~$ fprintd-enroll 
Using device /net/reactivated/Fprint/Device/0
Enrolling right-index-finger finger.
Enroll result: enroll-stage-passed
Enroll result: enroll-unknown-error
`

[image]

leroyorama, 2023-10-10:
Please see the link to an answer on askubuntu
[https://askubuntu.com/a/1488580/1737945](url)

jackfido, 2023-11-19:
> Please see the link to an answer on askubuntu
> 
> [https://askubuntu.com/a/1488580/1737945](url)

This works thank you!!!

[image](https://github.com/uunicorn/python-validity/assets/5467389/922f99cb-18d3-4b59-a158-15b4444e68e2)