uunicorn/python-validity #106: `open-fprintd-resume.service` fails to start after resuming from suspend

xyzshantaram, 2022-01-07:
It is fixed upon restarting, but fails upon resuming from suspend again. This is on **Arch Linux** with the **`python-validity-git`** AUR package, and my FPR is **`06cb:009a` Synaptics, Inc. Metallica MIS Touch Fingerprint Reader**.

Output of `systemctl status open-fprintd-resume.service`:
```
× open-fprintd-resume.service - Restart devices after resume
     Loaded: loaded (/usr/lib/systemd/system/open-fprintd-resume.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Fri 2022-01-07 13:12:24 IST; 13s ago
    Process: 38590 ExecStart=/usr/lib/open-fprintd/resume.py (code=exited, status=1/FAILURE)
   Main PID: 38590 (code=exited, status=1/FAILURE)
        CPU: 39ms

Jan 07 13:12:24 haruki resume.py[38590]:   File "/usr/lib/python3.10/site-packages/validitysensor/init_flash.py", line 113, in init_flash
Jan 07 13:12:24 haruki resume.py[38590]:     info = get_flash_info()
Jan 07 13:12:24 haruki resume.py[38590]:   File "/usr/lib/python3.10/site-packages/validitysensor/flash.py", line 41, in get_flash_info
Jan 07 13:12:24 haruki resume.py[38590]:     assert_status(rsp)
Jan 07 13:12:24 haruki resume.py[38590]:   File "/usr/lib/python3.10/site-packages/validitysensor/util.py", line 12, in assert_status
Jan 07 13:12:24 haruki resume.py[38590]:     raise Exception('Failed: %04x' % s)
Jan 07 13:12:24 haruki resume.py[38590]: Exception: Failed: 0315
Jan 07 13:12:24 haruki systemd[1]: open-fprintd-resume.service: Main process exited, code=exited, status=1/FAILURE
Jan 07 13:12:24 haruki systemd[1]: open-fprintd-resume.service: Failed with result 'exit-code'.
Jan 07 13:12:24 haruki systemd[1]: Failed to start Restart devices after resume.
```

Output of `sudo systemctl start open-fprintd-resume.service`:

```
Job for open-fprintd-resume.service failed because the control process exited with error code.
See "systemctl status open-fprintd-resume.service" and "journalctl -xeu open-fprintd-resume.service" for details.
```

Output of `journalctl -xeu open-fprintd-resume.service`:

```
Jan 07 13:12:24 haruki systemd[1]: open-fprintd-resume.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit open-fprintd-resume.service has entered the 'failed' state with result 'exit-code'.
Jan 07 13:12:24 haruki systemd[1]: Failed to start Restart devices after resume.
░░ Subject: A start job for unit open-fprintd-resume.service has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit open-fprintd-resume.service has finished with a failure.
░░ 
░░ The job identifier is 3350 and the job result is failed.
Jan 07 13:14:00 haruki systemd[1]: Starting Restart devices after resume...
░░ Subject: A start job for unit open-fprintd-resume.service has begun execution
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit open-fprintd-resume.service has begun execution.
░░ 
░░ The job identifier is 3432.
Jan 07 13:14:00 haruki resume.py[38872]: Traceback (most recent call last):
Jan 07 13:14:00 haruki resume.py[38872]:   File "/usr/lib/open-fprintd/resume.py", line 11, in <module>
Jan 07 13:14:00 haruki resume.py[38872]:     o.Resume()
Jan 07 13:14:00 haruki resume.py[38872]:   File "/usr/lib/python3.10/site-packages/dbus/proxies.py", line 141, in __call__
Jan 07 13:14:00 haruki resume.py[38872]:     return self._connection.call_blocking(self._named_service,
Jan 07 13:14:00 haruki resume.py[38872]:   File "/usr/lib/python3.10/site-packages/dbus/connection.py", line 652, in call_blocking
Jan 07 13:14:00 haruki resume.py[38872]:     reply_message = self.send_message_with_reply_and_block(
Jan 07 13:14:00 haruki resume.py[38872]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.22 was not provided by any .service files
Jan 07 13:14:00 haruki systemd[1]: open-fprintd-resume.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ An ExecStart= process belonging to unit open-fprintd-resume.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Jan 07 13:14:00 haruki systemd[1]: open-fprintd-resume.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit open-fprintd-resume.service has entered the 'failed' state with result 'exit-code'.
Jan 07 13:14:00 haruki systemd[1]: Failed to start Restart devices after resume.
░░ Subject: A start job for unit open-fprintd-resume.service has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit open-fprintd-resume.service has finished with a failure.
░░ 
░░ The job identifier is 3432 and the job result is failed.
```

I also tried it with the regular `python-validity` AUR package (which doesn't seem to have been updated to 0.13 yet?), but that also caused the service with a different error -- I'll post the details in a follow-up comment.

guemidiborhane, 2022-01-07:
how did you manage to fix this?

xyzshantaram, 2022-01-07:
I didn't, it's still broken.

guemidiborhane, 2022-01-07:
Sorry I miss understood

l4pa, 2022-01-09:
i can confirm the issue on fully upgraded Arch, libusb error, i suppose it has to do with upgraded python. reinstalled libusb, no effect. same model of fingerprint reader. 

> It is fixed upon restarting, but fails upon resuming from suspend again. This is on **Arch Linux** with the **`python-validity-git`** AUR package, and my FPR is **`06cb:009a` Synaptics, Inc. Metallica MIS Touch Fingerprint Reader**.
> 
> Output of `systemctl status open-fprintd-resume.service`:
> 
> ```
> × open-fprintd-resume.service - Restart devices after resume
>      Loaded: loaded (/usr/lib/systemd/system/open-fprintd-resume.service; enabled; vendor preset: disabled)
>      Active: failed (Result: exit-code) since Fri 2022-01-07 13:12:24 IST; 13s ago
>     Process: 38590 ExecStart=/usr/lib/open-fprintd/resume.py (code=exited, status=1/FAILURE)
>    Main PID: 38590 (code=exited, status=1/FAILURE)
>         CPU: 39ms
> 
> Jan 07 13:12:24 haruki resume.py[38590]:   File "/usr/lib/python3.10/site-packages/validitysensor/init_flash.py", line 113, in init_flash
> Jan 07 13:12:24 haruki resume.py[38590]:     info = get_flash_info()
> Jan 07 13:12:24 haruki resume.py[38590]:   File "/usr/lib/python3.10/site-packages/validitysensor/flash.py", line 41, in get_flash_info
> Jan 07 13:12:24 haruki resume.py[38590]:     assert_status(rsp)
> Jan 07 13:12:24 haruki resume.py[38590]:   File "/usr/lib/python3.10/site-packages/validitysensor/util.py", line 12, in assert_status
> Jan 07 13:12:24 haruki resume.py[38590]:     raise Exception('Failed: %04x' % s)
> Jan 07 13:12:24 haruki resume.py[38590]: Exception: Failed: 0315
> Jan 07 13:12:24 haruki systemd[1]: open-fprintd-resume.service: Main process exited, code=exited, status=1/FAILURE
> Jan 07 13:12:24 haruki systemd[1]: open-fprintd-resume.service: Failed with result 'exit-code'.
> Jan 07 13:12:24 haruki systemd[1]: Failed to start Restart devices after resume.
> ```
> 
> Output of `sudo systemctl start open-fprintd-resume.service`:
> 
> ```
> Job for open-fprintd-resume.service failed because the control process exited with error code.
> See "systemctl status open-fprintd-resume.service" and "journalctl -xeu open-fprintd-resume.service" for details.
> ```
> 
> Output of `journalctl -xeu open-fprintd-resume.service`:
> 
> ```
> Jan 07 13:12:24 haruki systemd[1]: open-fprintd-resume.service: Failed with result 'exit-code'.
> ░░ Subject: Unit failed
> ░░ Defined-By: systemd
> ░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
> ░░ 
> ░░ The unit open-fprintd-resume.service has entered the 'failed' state with result 'exit-code'.
> Jan 07 13:12:24 haruki systemd[1]: Failed to start Restart devices after resume.
> ░░ Subject: A start job for unit open-fprintd-resume.service has failed
> ░░ Defined-By: systemd
> ░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
> ░░ 
> ░░ A start job for unit open-fprintd-resume.service has finished with a failure.
> ░░ 
> ░░ The job identifier is 3350 and the job result is failed.
> Jan 07 13:14:00 haruki systemd[1]: Starting Restart devices after resume...
> ░░ Subject: A start job for unit open-fprintd-resume.service has begun execution
> ░░ Defined-By: systemd
> ░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
> ░░ 
> ░░ A start job for unit open-fprintd-resume.service has begun execution.
> ░░ 
> ░░ The job identifier is 3432.
> Jan 07 13:14:00 haruki resume.py[38872]: Traceback (most recent call last):
> Jan 07 13:14:00 haruki resume.py[38872]:   File "/usr/lib/open-fprintd/resume.py", line 11, in <module>
> Jan 07 13:14:00 haruki resume.py[38872]:     o.Resume()
> Jan 07 13:14:00 haruki resume.py[38872]:   File "/usr/lib/python3.10/site-packages/dbus/proxies.py", line 141, in __call__
> Jan 07 13:14:00 haruki resume.py[38872]:     return self._connection.call_blocking(self._named_service,
> Jan 07 13:14:00 haruki resume.py[38872]:   File "/usr/lib/python3.10/site-packages/dbus/connection.py", line 652, in call_blocking
> Jan 07 13:14:00 haruki resume.py[38872]:     reply_message = self.send_message_with_reply_and_block(
> Jan 07 13:14:00 haruki resume.py[38872]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.22 was not provided by any .service files
> Jan 07 13:14:00 haruki systemd[1]: open-fprintd-resume.service: Main process exited, code=exited, status=1/FAILURE
> ░░ Subject: Unit process exited
> ░░ Defined-By: systemd
> ░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
> ░░ 
> ░░ An ExecStart= process belonging to unit open-fprintd-resume.service has exited.
> ░░ 
> ░░ The process' exit code is 'exited' and its exit status is 1.
> Jan 07 13:14:00 haruki systemd[1]: open-fprintd-resume.service: Failed with result 'exit-code'.
> ░░ Subject: Unit failed
> ░░ Defined-By: systemd
> ░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
> ░░ 
> ░░ The unit open-fprintd-resume.service has entered the 'failed' state with result 'exit-code'.
> Jan 07 13:14:00 haruki systemd[1]: Failed to start Restart devices after resume.
> ░░ Subject: A start job for unit open-fprintd-resume.service has failed
> ░░ Defined-By: systemd
> ░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
> ░░ 
> ░░ A start job for unit open-fprintd-resume.service has finished with a failure.
> ░░ 
> ░░ The job identifier is 3432 and the job result is failed.
> ```
> 
> ~I also tried it with the regular `python-validity` AUR package (which doesn't seem to have been updated to 0.13 yet?), but that also caused the service with a different error -- I'll post the details in a follow-up comment.~
> 
> Edit: I'm unable to reproduce that error anymore, it was a USBError from pyusb stating that the device couldn't be found and that it may have been disconnected. Now, both the `-git` and normal versions report the same error.

l4pa, 2022-01-09:
it only gets fixed for me if both python3-validity and open-fprintd are restarted.

l4pa, 2022-01-12:
my log of open-fprintd-resume failure just in case

`× open-fprintd-resume.service - Restart devices after resume
     Loaded: loaded (/usr/lib/systemd/system/open-fprintd-resume.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Wed 2022-01-12 22:14:20 EET; 8s ago
    Process: 4606 ExecStart=/usr/lib/open-fprintd/resume.py (code=exited, status=1/FAILURE)
   Main PID: 4606 (code=exited, status=1/FAILURE)
        CPU: 46ms

Jan 12 22:14:20 zzz resume.py[4606]:   File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 837, in bulk_write
Jan 12 22:14:20 zzz resume.py[4606]:     return self.__write(self.lib.libusb_bulk_transfer,
Jan 12 22:14:20 zzz resume.py[4606]:   File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 938, in __write
Jan 12 22:14:20 zzz resume.py[4606]:     _check(retval)
Jan 12 22:14:20 zzz resume.py[4606]:   File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 604, in _check
Jan 12 22:14:20 zzz resume.py[4606]:     raise USBError(_strerror(ret), ret, _libusb_errno[ret])
Jan 12 22:14:20 zzz resume.py[4606]: usb.core.USBError: [Errno 19] No such device (it may have been disconnected)
Jan 12 22:14:20 zzz systemd[1]: open-fprintd-resume.service: Main process exited, code=exited, status=1/FAILURE
Jan 12 22:14:20 zzz systemd[1]: open-fprintd-resume.service: Failed with result 'exit-code'.
Jan 12 22:14:20 zzz systemd[1]: Failed to start Restart devices after resume.
`

sarlej, 2022-01-18:
i have same issue, what helps for me as workaround (instead of restarting pc) is restarting open-fprintd and python3-validity
```
sudo systemctl restart open-fprintd
sudo systemctl restart python3-validity
```

ioanmoldovan95, 2022-01-22:
I have the same issue as the one described above. 
I have solved it by writing a service that runs after resume and restarts open-fprintd and python3-validity services
It's a quick and easy fix until a permanent fix is found
I'll leave the code for this service below for anyone who might need it.

`[Unit]
Description=Restart services to fix fingerprint integration
After=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target

[Service]
Type=oneshot
ExecStart=systemctl restart open-fprintd.service python3-validity.service

[Install]
WantedBy=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
`

xyzshantaram, 2022-01-23:
@ioanmoldovan95 Thanks! I was considering doing something like that myself, you're a lifesaver! <3

Arsur, 2022-05-13:
I am still having this error. If iam not using  my own script i get stuck on my login screen. With the service from [ioanmoldovan95](https://github.com/ioanmoldovan95) i can login by password.

saunite, 2022-05-16:
> I have the same issue as the one described above. I have solved it by writing a service that runs after resume and restarts open-fprintd and python3-validity services It's a quick and easy fix until [...]

This worked perfectly for me on a Lenovo T480s running Fedora 36

IceCloud76, 2022-07-17:
> > I have the same issue as the one described above. I have solved it by writing a service that runs after resume and restarts open-fprintd and python3-validity services It's a quick and easy fix until a permanent fix is found I'll leave the code for this service below for anyone who might need it.
> > ```
> > [Unit]
> > Description=Restart services to fix fingerprint integration
> > After=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
> > 
> > [Service]
> > Type=oneshot
> > ExecStart=systemctl restart open-fprintd.service python3-validity.service
> > 
> > [Install]
> > WantedBy=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
> > ```
> 
> Just tried this on my T480 running Fedora 36. Works like a charm!

Also running Fedora 36 with a T480, although I've noticed that at least with KDE plasma, only about half the time I'm prompted for a fingerprint. The other times I need to put in a bad password once before it prompts  for a fingerprint. Maybe the services don't restart fast enough for the device to register?

SashaPGT, 2022-10-15:
> I have the same issue as the one described above. I have solved it by writing a service that runs after resume and restarts open-fprintd and python3-validity services It's a quick and easy fix until [...]

This worked great on my T480s running Fedora 37 Beta, thanks!

perezandrew477, 2022-11-04:
> > > I have the same issue as the one described above. I have solved it by writing a service that runs after resume and restarts open-fprintd and python3-validity services It's a quick and easy fix until a permanent fix is found I'll leave the code for this service below for anyone who might need it.
> > > ```
> > > [Unit]
> > > Description=Restart services to fix fingerprint integration
> > > After=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
> > > 
> > > [Service]
> > > Type=oneshot
> > > ExecStart=systemctl restart open-fprintd.service python3-validity.service
> > > 
> > > [Install]
> > > WantedBy=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
> > > ```
> > 
> > 
> > Just tried this on my T480 running Fedora 36. Works like a charm!
> 
> Also running Fedora 36 with a T480, although I've noticed that at least with KDE plasma, only about half the time I'm prompted for a fingerprint. The other times I need to put in a bad password once before it prompts for a fingerprint. Maybe the services don't restart fast enough for the device to register?
> 
> Edi: It does seem like a timing thing. Waiting a second or two before hitting the enter key on the lock screen makes it so the fingerprint reader always comes up. But because I'm always faster than that 2 second wait, it doesn't prompt on first try :(

Tried this fix as well and I'm experiencing the same behavior. If I get pass the lock screen too quickly, I won't be able to use the fingerprint reader to login. If I wait a second or two, it works just fine and I can login with my fingerprint. I don't really mind waiting a second or two, but hopefully this can be improved upon. Thanks @ioanmoldovan95!

Grafcube, 2023-01-10:
I fixed this issue for myself with the solution in this thread and it worked perfectly on my Thinkpad running latest Arch. Is there any particular reason that this fix hasn't been included in the main package?

ioanmoldovan95, 2023-02-09:
Yes, I am aware about the issue that is described in the last few comments, but haven't found a solution to it yet as it doesn't bother me too much, I prefer to wait 1-2 seconds before tapping enter to bring up the login screen. It is for sure a timing issue, the services are not restarted as fast as they should. If I come up with a solution I will post it here

coignard, 2024-03-05:
You can also enable services that come with the python-validity package instead of creating a new one:

`systemctl enable python3-validity-restart-after-resume.service open-fprintd-restart-after-resume.service'.

cg505, 2024-03-11:
@coignard What package are you using? I don't see those service files in this repo or in the AUR packages.

coignard, 2024-03-11:
> @coignard What package are you using? I don't see those service files in this repo or in the AUR packages.

`tigro/python-validity` copr on Fedora.

poopsicles, 2024-08-07:
simply restarting `python3-validity` manually allows me to use fingerprint auth (with `fprintd-verify` for example) after unlocking from suspend, and enabling `python3-validity-restart-after-resume` lets it work on the lock screen, with a caveat.

if you get past the lock screen clock in GNOME, the service isn't restarted in time which allows you to only use your password to log in (as shown in earlier comments). however, for some reason, my install of fedora *skips* the clock (i can see it flash by really quick), thereby dumping my at the password entry...after logging in though, `fprintd-verify` works fine.

This [Gitlab issue](https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/7791) also seems to be about the same thing, and as a workaround tapping Esc (to go back to the clock) and then Enter again loads it correctly.

For reference, I'm using a ThinkPad T480s and Fedora 40 (but it's a mostly fresh install, haven't gotten around to `sudo dnf update`ing, maybe that would fix the problem.

meandyc, 2024-08-26:
Works in Debian 12 (T480) too.

G1Sar7, 2025-06-06:
Hello everyone!
I don't really know if this can help but it worked for me (Thinkpad T480 with Arch).

https://gist.github.com/borcean/f32c47f6cc52cee33dfc2265ce63f777

Try to follow the installation for python-validity from AUR repo, install all the packages it asks for and then, as soon as the issue pops in the terminal follow the link I shared above.

The steps are for fedora but they are applicable to Arch as well.
If you have troubles you can also ask ChatGPT for the correct services activation and edits to the files for the OS to ask you for the fingerprint.

It worked for me on a clean new install of the OS no issues at all, even after the problem popped on my console.

Hope this helped! :D

stdio44, 2026-10-20:
I also have the 06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader, and ran into the same problem after a recent update to Ubuntu 26.04.1, on my ThinkPad P72. I guess there are multiple ways to approach the same problem, but I just updated `open-fprintd-resume.service` with a single line fix (the `ExecStartPre`):

```
[Unit]
Description=Restart devices after resume
After=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target

[Service]
Type=oneshot
ExecStartPre=/usr/bin/systemctl restart python3-validity
ExecStart=/usr/lib/open-fprintd/resume.py

[Install]
WantedBy=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
```