uunicorn/python-validity #107: Cannot enable python3-validity

hasmar04, 2022-01-23:
When trying to enable python3-validity via `sudo systemctl enable python3-validity` on Manjaro the following error occurs:
`The unit files have no installation config (WantedBy=, RequiredBy=, Also=,
Alias= settings in the [Install] section, and DefaultInstance= for template
units). This means they are not meant to be enabled using systemctl.
 
Possible reasons for having this kind of units are:
• A unit may be statically enabled by being symlinked from another unit's
  .wants/ or .requires/ directory.
• A unit's purpose may be to act as a helper for some other unit which has
  a requirement dependency on it.
• A unit may be started when needed via activation (socket, path, timer,
  D-Bus, udev, scripted systemctl call, ...).
• In case of template units, the unit is meant to be enabled with some
  instance name specified.`

LukeFernandes, 2022-02-08:
I get the same error! Also on Manjaro, kernel 5.10.

lululock71, 2022-04-24:
Same under Arch 5.17.4-zen.