uunicorn/python-validity #111: Getting enroll-failed on Thinkpad t570

njakes, 2022-02-23:
First off, using your setup is the first time I have gotten the fingerprint reader to activate on linux.  I initially downoaded it on Arch and the very first enroll went ok.  Then I tried to utilize the new plasma 5.24 integrated fingerprint unlock in settings.  Ever since then it has started failing to enroll.  I tried deleting the first enroll using fprintd.  Now I cant even enroll any fingerprints even after switching to Zorin OS. When checking on the service I get:
```
04c3
Traceback (most recent call last):
 File "/usr/lib/python-validity/dbus-service", line 160, in run sensor.enroll(usr, index, update_cb)
File "/usr/lib/python3/dist-packages/validitysensor/sensor.py", line 853, in enroll return do_create_finger(template, tid)
File "/usr/lib/python3/dist-packages/validitysensor/sensor.py", line 819, in do_cr> recid = db.new_finger(usr, tinfo)
File "/usr/lib/python3/dist-packages/validitysensor/db.py", line 235, in new_finger rec = self.new_record(userid, 0xb, stg.dbid, template)
File "/usr/lib/python3/dist-packages/validitysensor/db.py", line 218, in new_record assert_status(rsp)
File "/usr/lib/python3/dist-packages/validitysensor/util.py", line 12, in assert_s> raise Exception('Failed: %04x' % s)
Exception: Failed: 04c3
```
That same error showed up in both arch and ubuntu based zorin.

uunicorn, 2022-02-23:
Hi, it's been ages since I've looked at this project, but IIRC, 04c3 means a "duplicate record". 
You are probably trying to add a record with the same finger id, or something. Try doing a factory reset.

njakes, 2022-02-24:
Ok thanks for the advice. Unfortunately the factory reset doesn't work either. Do you know of an active project similar to yours?  I don't want to cause any undue stress if this project isn't very high on your list.