uunicorn/python-validity #115: Fingerprint reader stops working on suspend

kozross, 2022-03-17:
I've set up `python-validity` according to the instructions for my distro (Arch Linux) and enrolled a fingerprint successfully. This works, but only until I first suspend. Once this happens, upon waking, I can no longer log in via fingerprint scan (the option isn't even offered), and when I check the status of `fprintd.service`, I get this:

```
[koz@Emi ~]$ systemctl status fprintd.service
○ fprintd.service - Fingerprint Authentication Daemon
     Loaded: error (Reason: Unit fprintd.service failed to load properly, please adjust/correct and reload service manager: File exists)
     Active: inactive (dead)
       Docs: man:fprintd(1)

Mar 17 16:45:57 Emi systemd[1]: fprintd.service: Two services allocated for the same bus name net.reactivated.Fprint, refusing operation.
```

Enabling the [suspend-related services](https://github.com/uunicorn/python-validity#fingerprint-not-working-after-waking-up-from-suspend) does not help. I do not have any other `fprintd`-related anything installed: only what was required for the AUR package. I'm using a ThinkPad T580, with `python-validity 0.12-5` according to `pacman -Qs`. My DE is GNOME.

piotrzientarski, 2022-04-18:
I have the same problem on Fedora 36.

Arsur, 2022-05-07:
I have the same issue on the Thinkpad Carbon X1 gen6, but this solution does not work for me. Any other tips?
fprintd.service unable to start

mjzjz, 2023-06-15:
> 
I have the same issue on the Thinkpad Carbon X1 gen6, too.How did you fix it?

gamecraftCZ, 2024-01-21:
Same problem here with ThinkPad X1 Yoga gen 4.
After resuming from suspend I can still se my device in lsusb: `Bus 001 Device 006: ID 06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader`
However I get this error from open-fprintd-resume service. 
```
Jan 21 18:11:08 NoYoga resume.py[5836]: usb.core.USBError: [Errno 19] No such device (it may have been disconnected)
Jan 21 18:11:08 NoYoga systemd[1]: open-fprintd-resume.service: Main process exited, code=exited, status=1/FAILURE
Jan 21 18:11:08 NoYoga systemd[1]: open-fprintd-resume.service: Failed with result 'exit-code'.
Jan 21 18:11:08 NoYoga systemd[1]: Failed to start Restart devices after resume.
```

gamecraftCZ, 2024-01-21:
As for workaround you can try this: https://github.com/uunicorn/python-validity/issues/106#issuecomment-1019342483

YovanggaAnandhika, 2024-10-25:
solutions and explanations that can be tried https://github.com/uunicorn/python-validity/issues/128#issuecomment-2437045697