uunicorn/python-validity #116: Finger names supplied to ListEnrolledFingers in DBus API are incorrect

espidev, 2022-03-28:
Hi,
It seems when ListEnrolledFingers is called, we are getting finger names such as "WINBIO_ANSI_381_POS_RH_INDEX_FINGER". The fprint API spec specifies that finger names should be of the following form: https://fprint.freedesktop.org/fprintd-dev/Device.html#fingerprint-names. So mapped to "right-index-finger" instead. 

Downstream issue, fingers not listing in Plasma's fingerprint dialog: https://bugs.kde.org/show_bug.cgi?id=450859

baschi29, 2022-04-16:
I took a quick look at the source code and this seems to be intentional to comply with the microsoft specification: https://docs.microsoft.com/en-us/windows/win32/secbiomet/winbio-ansi-381-pos-fingerprint-constants

Relevant Code: https://github.com/uunicorn/python-validity/blob/master/dbus_service/dbus-service#L143-L147

I don't know if this is neccessary for the sensors to function or if this was choosen for windows interopability. As a fix maybe it would be possible to enroll a given fingerprint under both names? Or to have a way to configure which option to choose?

espidev, 2022-04-16:
I mean sure it can be used internally but this is the dbus api we're talking about, which is used by applications.

PR #120 (2022-04-20): Comply with fprint API when returning enrolled fingers
The ANSI381 names implemented with #23 did not comply with the fprint API spec.

This PR fixes #116 by substituting all ANSI381 names when there is a corresponding fprint name available. For backwards compatibility ANSI381 naming is still used for everything else.
closed 2022-04-20 by uunicorn