uunicorn/python-validity #117: Validity Sensors, Inc. VFS495 Fingerprint Reader not supported ?

NoxInmortus, 2022-04-01:
Hello, anyone able to confirm if `Validity Sensors, Inc. VFS495 Fingerprint Reader` is supported or not ?

```
lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04f2:b530 Chicony Electronics Co., Ltd HP HD Camera
Bus 001 Device 003: ID 138a:003f Validity Sensors, Inc. VFS495 Fingerprint Reader
Bus 001 Device 002: ID 8087:0a2b Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
● python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/lib/systemd/system/python3-validity.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-04-01 11:37:40 CEST; 1min 43s ago
    Process: 11190 ExecStart=/usr/lib/python-validity/dbus-service --debug (code=exited, status=1/FAILURE)
   Main PID: 11190 (code=exited, status=1/FAILURE)

Apr 01 11:37:40 aeg-ow2 dbus-service[11190]:     init.open()
Apr 01 11:37:40 aeg-ow2 dbus-service[11190]:   File "/usr/lib/python3/dist-packages/validitysensor/init.py", line 47, in open
Apr 01 11:37:40 aeg-ow2 dbus-service[11190]:     usb.open()
Apr 01 11:37:40 aeg-ow2 dbus-service[11190]:   File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 50, in open
Apr 01 11:37:40 aeg-ow2 dbus-service[11190]:     self.open_dev(dev)
Apr 01 11:37:40 aeg-ow2 dbus-service[11190]:   File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 62, in open_dev
Apr 01 11:37:40 aeg-ow2 dbus-service[11190]:     raise Exception('No matching devices found')
Apr 01 11:37:40 aeg-ow2 dbus-service[11190]: Exception: No matching devices found
Apr 01 11:37:40 aeg-ow2 systemd[1]: ^[[0;1;39m^[[0;1;39mpython3-validity.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 11:37:40 aeg-ow2 systemd[1]: ^[[0;1;38;5;185m^[[0;1;39m^[[0;1;38;5;185mpython3-validity.service: Failed with result 'exit-code'.
```

alemoh1234, 2022-05-06:
I have the same sensor and had the same problem, it seems it really isn`t supported

nikp123, 2022-07-19:
It's not supported, but I'd like to hear if any of the maintainers/contributors can decode this wireshark dump.

It was done on a Windows 10 LTSC 20H2 machine with the validity drivers pulled automatically from Windows update.
The driver package *seems* incompatible with the existing reverse engineering tools that are provided on this GitHub account, since the dll files are completely different.

The first dump is just the verification of the first fingerprint (right index finger). Does not include USBHUB info because that's only done during bootup.

[cap1.pcapng.zip](https://github.com/uunicorn/python-validity/files/9144158/cap1.pcapng.zip)

While the other dump contains the whole USB sequence from bootup (starting at about 20s into the dump) + verification of the same fingerprint.

[cap2.pcapng.zip](https://github.com/uunicorn/python-validity/files/9144160/cap2.pcapng.zip)

I'd like to hear if anyone can decode what any of those USB In/Out packets mean. They're probably some stupidly proprietary data protocol.

nikp123, 2022-07-19:
If the exact driver executable is needed for further research, I'm happy to provide.

nagaisa, 2023-01-05:
im pretty sure [this](https://ftp.ext.hp.com/pub/softpaq/sp84501-85000/sp84530.tar) is the proprietary from the vendor if this can help