uunicorn/python-validity #118: Unable to start python3-validity service || Exception: No supported validity device found

LeonardoMantovani, 2022-04-04:
I've seen [a video on YouTube](https://www.youtube.com/watch?v=4rWYXc4F-jU) that was showing how to enable fingerprint support on Linux even for [fingerprint readers not supported by `fprint`](https://fprint.freedesktop.org/supported-devices.html), thanks to this package.

I tried installing the package on Fedora 35 as described in the ReadMe but, after `fprintd-enroll` I got the following error:
> Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice:

So, as the ReadMe suggests, I checked the status of `python3-validity` and I saw it was `inactive (dead)`.
Continuing to follow the instructions, I tried to factory reset my fingerprint sensor but with the command `sudo validity-sensors-firmware` I got this output:
> Traceback (most recent call last):
  File "/usr/bin/validity-sensors-firmware", line 92, in <module>
    raise Exception('No supported validity device found')
Exception: No supported validity device found

What am I doing wrong? Maybe the video was lying, and this package don't support all fingerprint sensors?
And, in that case, is there a way I can help to have my sensor supported?

*(In case it could be useful, this is the information I got with `lsusb` about my fingerprint sensor):*
> Bus 003 Device 004: ID 27c6:6094 Shenzhen Goodix Technology Co.,Ltd. Goodix USB2.0 MISC

physicophilic, 2022-05-14:
Exactly same situation!

argibbs, 2022-05-18:
It's (probably) not the sensors.

I got this working on Ubuntu straight out of the box, then tried Fedora, and hit the same issue.

Looks like in the copr repo, the install is not quite right, specifically you're missing a couple of files. Once they exist, the app will populate them, but it blows up if it can't find them.

See https://www.mark-gilbert.co.uk/validity-fingerprint-sensor-06cb009a-on-lenovo-x1-carbon-running-fedora/, specifically:
> Create two missing files that don’t get created
> `$ sudo touch /usr/share/python-validity/backoff && sudo touch /usr/share/python-validity/calib-data.bin`

This didn't completely solve my problem; fprintd-enroll was working, but the sensor wasn't firing and the journalctl logs showed lots of timeouts.

So I reset the firmware as described in the link above ☝️ and in the troubleshooting section in this project's README. And that fixed it.

saadahmsiddiqui, 2026-03-25:
Facing the same issue on thinkpad l380