uunicorn/python-validity #119: Can't enroll my finger

mircsicz, 2022-04-09:
Hi all,

first of all let me say thanks for giving me hope... the fingerprint reader is the only thing I'm missing on my machine running fedora on my X1 Yoga 3rd.

I've found the project throught this reddit post [ThinkPad Synaptics Fingerprint-Reader](https://www.reddit.com/r/thinkpad/comments/ihotav/linux_fingerprint_sensor_support_for_synaptics/)

I'm on FedoraCore35 running 5.16.18 and sadly ran into problem's setting up the fingerprint-reader:
I followed the readme but the reader wasn't detected, so I googled and found that [posting](https://github.com/uunicorn/python-validity/issues/103#issuecomment-966016605)

With that I came a step closer, now fingerprint-enroll allowed me to enroll, but it would never finish the enrollment: And I could see that the reader has a LED on top of it. But that LED barely shows activity... Seems to only turn on for a slight moment when I start enrollment and in the moment I Ctrl-C finish it...

Now when I try to enroll I see issue's like this:
`
[mircsicz@x1-yoga ~]$ fprintd-enroll 
Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: 
[mircsicz@x1-yoga ~]$ fprintd-enroll 
Using device /net/reactivated/Fprint/Device/2
Enrolling right-index-finger finger.
Enroll result: enroll-failed
EnrollStop failed: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Remote peer disconnected
[mircsicz@x1-yoga ~]$ fprintd-enroll 
Using device /net/reactivated/Fprint/Device/3
Enrolling right-index-finger finger.
^C
`

So from three to five try's one might start to enroll, but as written above never finishes...

Here's an output from journalctl:
`
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:In EnrollStart right-index-finger for mirco
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:>tls> 17: 6901000000
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:>cmd> 1703030040d309f04e9622ce12efa3ff3cc64fe1f9b5d822ef2fe6fcf780e52a162b15232d6303d2480bfca4ac98e21ee19b9c3>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<cmd< 1703030040f4c5d7383cc79df61a599a0b5fcee808e4e991f9f6a1bb8a30abace0581a8af176b356cf0ca0bd70d78b35a3b7d30>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<tls< 17: 0000
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:>tls> 17: 3920bf0200ffff0000019900200000000099990000000000000000000000000020000000000000000000000000ffff00000>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:>cmd> 17030300b0c8aad12e12e5f027038ca0941f0889cb009d4571e6e908f4325f968dd37f27573a245fb4aaae5428c1c3662632494>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<cmd< 17030300405b3e30d5b7ec63b3739c91fe49cdbaeef71ea5508eb8ac4f4c33625226351a49774c698844ad7fb57bec023b2cf71>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<tls< 17: 0000
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:>tls> 17: 027800000023000000200008000020008000000100320074000000008020200400242000005020773628200100302001003>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:>cmd> 1703030400c913f9b664fcd54e3f9d9384c9318c71eb09370aa5f2ebfa009132d7f26ed0978bbccd9cf451a62311a456da4ba02>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<cmd< 17030307e0c8aa75fe338814e33af0c6817d40db5475b384c4e8a5f04f5ecacb3e2f7f1d304dab05c47cb2bb9b7080eaa8014f3>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<tls< 17: 0000880200305e0000000000000000000000000000000000000000000000000000000000000000000000000000000000000>
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<int< 0000000000
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<int< 0200401000
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<int< 0340010000
Apr 09 11:36:04 x1-yoga dbus-service[8201]: DEBUG:root:<int< 0341030040
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:>tls> 17: 04
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:>cmd> 1703030040401d29e55c3a726b492c2c999667e31893f50bbe853eb3c5d5f5a64f5d954b0845639d315f9404ae0107487f34970>
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:<cmd< 17030300407d1c2a96e6ea2c0d966a9c675d252a050aa25cc3b5f24570fb221e9b53ab05e4315e8133ba1027847430c59dd6e15>
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:<tls< 17: 0000
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:>tls> 17: 39f4010000f401000001ff002000000000ffff0000000000000000000000000020000000000000000000000000f40100000>
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:>cmd> 17030300b07feb35fbf96081578d599ae87f053a0e013aa8273f4ede5fbae39cf2364dbbc47dc3b95dfe6f2a303bb53cda4ad29>
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:<cmd< 17030300408aa2f215e4abb2d82cd73a9c3e475bb03807cdca41df8fb7a4615a7d8aef9750ec418beb2dd6224ed1d44a3650d34>
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:<tls< 17: 0000
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:>tls> 17: 6900000000
Apr 09 11:36:29 x1-yoga dbus-service[8201]: DEBUG:root:>cmd> 1703030040e7414742802eec52fb6bbed01e0f90a945e74fd45bae5b61796fa7f162157597f7289e798e2651b7083616c91678e>
Apr 09 11:36:30 x1-yoga dbus-service[8201]: DEBUG:root:<cmd< 1703030040df888d9e5b8c09bc4f3c68e94c53402cc2e170f4019779a8ef07f0cc82538dacab28aa4bf54876ec1ca4f7d169d00>
Apr 09 11:36:30 x1-yoga dbus-service[8201]: DEBUG:root:<tls< 17: 0000
Apr 09 11:36:30 x1-yoga ERROR:root[8201]: Traceback (most recent call last):
                                            File "/usr/lib/python3.10/site-packages/validitysensor/usb.py", line 128, in wait_int
                                              resp = self.dev.read(131, 1024, timeout=100)
                                            File "/usr/lib/python3.10/site-packages/usb/core.py", line 1029, in read
                                              ret = fn(
                                            File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 864, in intr_read
                                              return self.__read(self.lib.libusb_interrupt_transfer,
                                            File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 954, in __read
                                              _check(retval)
                                            File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 602, in _check
                                              raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
                                          usb.core.USBTimeoutError: [Errno 110] Operation timed out
                                          
                                          During handling of the above exception, another exception occurred:
                                          
                                          Traceback (most recent call last):
                                            File "/usr/lib/python-validity/dbus-service", line 161, in run
                                              sensor.enroll(usr, index, update_cb)
                                            File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 844, in enroll
                                              raise e
                                            File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 832, in enroll
                                              self.capture(CaptureMode.ENROLL)
                                            File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 711, in capture
                                              b = usb.wait_int()
                                            File "/usr/lib/python3.10/site-packages/validitysensor/usb.py", line 135, in wait_int
                                              raise CancelledException()
                                          validitysensor.usb.CancelledException
Apr 09 11:36:30 x1-yoga dbus-service[8201]: DEBUG:root:EnrollStatus
`

And here I'm officially lost! Hope one of you has a hint for me...

Lapazlane, 2022-06-03:
Have you tried deleting fingerprint data in the BIOS first (security/fingerprint tab)?

cheetahdotcat, 2023-05-02:
> Have you tried deleting fingerprint data in the BIOS first (security/fingerprint tab)?

that solved it for me