uunicorn/python-validity #122: Support Ubuntu 22.04?

ankworld, 2022-04-26:
Can you publish PPA for jammy?

rebelwolfson, 2022-05-04:
Meanwhile, one could try `sudo apt-get update --allow-insecure-repositories` (it is insecure, as you could gather)

ikhsangama, 2022-05-13:
yes pleaseee

Lapazlane, 2022-06-02:
Hello, this enables my Thinkpad Yoga 370 to use fully use fingerprint scanner. It would be brilliant if you could update for Ubuntu (etc) 22.04 LTS? I'm holding off updating until I can use python-validity. Thanks you for your  great work on this.

uunicorn, 2022-06-02:
Will do. It would be nice to get #127 fixed first, though.

Lapazlane, 2022-06-02:
Fantastic! That's great news.

uunicorn, 2022-06-02:
PPA published. 
I haven't done any testing on Jammy though.

Lapazlane, 2022-06-02:
That's great news, thank you for the very prompt response.. I'll be celebrating Her Majesty's platinum celebration this weekend, so I'll update to Jammy next week and report back.

Lapazlane, 2022-06-03:
Confirming working on Thinkpad Yoga 370 on Ubuntu 22.04 Jammy Jellyfish, using live USB, on a dual booting with windows 10 set up, following the instructions. I did have to manually download python3-usb and innoextract packages first, to overcome the dependency missing alert. 

So I can log in using fingerprint. Not run pam-auth-update yet, to check  for sudo authorisation, but no reason to expect any issues. Feel confident to do a full install/update to 22.04.

Side note - having just fresh installed ubuntu 21.10, I was having problems getting fingerprint to enroll. I fixed by going into initial BIOS, selct security/finger print and erase fingerprint data. This does seem to mean that I can only enroll to either Linux or Windows at any one time, but as I don't actually use the windows install, this isn't a problem for me.

uunicorn, 2022-06-03:
> Side note - having just fresh installed ubuntu 21.10, I was having problems getting fingerprint to enroll. I fixed by going into initial BIOS, selct security/finger print and erase fingerprint data. [...]

This is odd. Interoperability supposed to work, as long as python-validity can access `/sys/class/dmi/id/product_serial` and `/sys/class/dmi/id/product_name` files.

lckarssen, 2022-08-01:
Indeed, I too can confirm this works on Ubuntu 22.04.

closed 2022-08-03 by ankworld